u236/homed-service-zigbee #3: Добавить топик статуса

kerenskiy, 2022-10-19:
Добавить топик статуса (service/status=online) для отслеживания работы сервиса.

u236, 2022-10-19:
Это реализовано в версии <code>3.0.37</code>, данные публикуются в топик <code>homed/service/zigbee</code>.
При старте приложения:
```json
{
  "started":"2022-10-19T10:10:43.988","uptime":0}
```

u236, 2022-10-19:
В следующем релизе добавлю MQTT LWT фичи.

u236, 2022-10-19:
Реализовал MQTT LWT в https://github.com/u236/homed-service-common/commit/e27bd210bdd9bf7ef276417e7d53c0d90cc4a55b, сообщение сокращено.

При старте приложения:
```json
{
  "status": "online"
}
```

При отсановке/падении приложения или потере соединения с брокером:
```json
{
  "status": "offline"
}
```

closed 2022-10-19 by u236