u236/homed-service-zigbee #10: Relay Sonoff ZB mini

d51x, 2022-10-26:
2022.10.26 16:24:58.557 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" joined network with address "0x4DA5"
2022.10.26 16:24:58.635 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:24:58.638 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:25:08.958 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview aborted: "node descriptor request failed"
2022.10.26 16:25:09.100 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:25:09.170 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:25:18.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview aborted: "endpoint 0x01 simple descriptor request failed"
2022.10.26 16:25:19.148 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:25:23.317 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" left network
2022.10.26 16:25:29.850 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" joined network with address "0x3555"
2022.10.26 16:25:29.931 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:25:29.934 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:25:30.011 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:25:30.087 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:25:30.223 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:25:33.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out
2022.10.26 16:25:44.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out

d51x, 2022-10-26:
2022.10.26 16:27:29.721 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" left network
2022.10.26 16:27:35.989 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" joined network with address "0xA379"
2022.10.26 16:27:35.992 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" rejoined network with address "0xA379"
2022.10.26 16:27:36.098 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:27:36.101 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:27:36.104 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:27:36.107 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:27:36.182 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:27:36.185 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:27:36.274 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:27:36.277 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:27:36.402 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:27:36.405 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:27:51.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out
2022.10.26 16:27:51.960 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out
2022.10.26 16:27:55.942 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" left network
2022.10.26 16:28:07.028 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" joined network with address "0x63B0"
2022.10.26 16:28:07.032 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" rejoined network with address "0x63B0"
2022.10.26 16:28:07.121 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:28:07.124 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:28:07.127 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" is router
2022.10.26 16:28:07.131 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" node descriptor received, manufacturer code: "0x1286"
2022.10.26 16:28:07.203 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:28:07.206 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" active endpoints received: "0x01, 0xF2"
2022.10.26 16:28:07.283 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:28:07.286 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0x01" simple descriptor received
2022.10.26 16:28:07.416 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:28:07.419 (inf) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" endpoint "0xF2" simple descriptor received
2022.10.26 16:28:17.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview aborted: "read basic attributes request failed"
2022.10.26 16:28:32.957 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out
2022.10.26 16:28:32.960 (wrn) zigbee:    Device "00:12:4b:00:24:c0:2a:d7" interview timed out

d51x, 2022-10-26:
не джойнится

u236, 2022-10-31:
Проблема с интервтю решена в https://github.com/u236/homed-service-zigbee/commit/9afa18acb783cc5cae014c48d0c7c01c2e5915d8.