u236/homed-service-zigbee #12: Relay Tuya TS0001 (_TZ3000_rmjr4ufz)

d51x, 2022-10-26:
2022.10.26 16:44:24.674 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" joined network with address "0xA2E1"
2022.10.26 16:44:24.677 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" rejoined network with address "0xA2E1"
2022.10.26 16:44:24.764 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" is router
2022.10.26 16:44:24.767 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" node descriptor received, manufacturer code: "0x1141"
2022.10.26 16:44:24.771 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" is router
2022.10.26 16:44:24.774 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" node descriptor received, manufacturer code: "0x1141"
2022.10.26 16:44:24.844 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" active endpoints received: "0x01, 0xF2"
2022.10.26 16:44:24.847 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" active endpoints received: "0x01, 0xF2"
2022.10.26 16:44:24.922 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" simple descriptor received
2022.10.26 16:44:24.925 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" simple descriptor received
2022.10.26 16:44:25.003 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" endpoint "0xF2" simple descriptor received
2022.10.26 16:44:25.008 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" endpoint "0xF2" simple descriptor received
2022.10.26 16:44:25.090 (wrn) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" manufacturer name "_TZ3000_rmjr4ufz" unrecognized
2022.10.26 16:44:25.094 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" manufacturer name is "_TZ3000_rmjr4ufz" and model name is "TS0001"
2022.10.26 16:44:25.097 (inf) zigbee:    Device "a4:c1:38:6a:08:33:da:bc" interview finished
2022.10.26 16:44:31.175 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" attribute "0xD003" with data type "0x42" and data "41:41:41:41"
2022.10.26 16:44:31.178 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" attribute "0xD003" with data type "0x42" and data "41:41:41:41"
2022.10.26 16:44:31.266 (wrn) zigbee:    Unrecognized attribute "0xD001" data type "0x48" received from device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" with payload: "02:00:06"
2022.10.26 16:44:31.272 (wrn) zigbee:    Unrecognized attribute "0xD001" data type "0x48" received from device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" with payload: "02:00:06"
2022.10.26 16:44:31.362 (wrn) zigbee:    Unrecognized attribute "0xD002" data type "0x48" received from device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" with payload: "02:00:0a"
2022.10.26 16:44:31.367 (wrn) zigbee:    Unrecognized attribute "0xD002" data type "0x48" received from device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE000" with payload: "02:00:0a"
2022.10.26 16:44:31.448 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:31.454 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x4001" with data type "0x21" and data "00:00"
2022.10.26 16:44:31.457 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x4002" with data type "0x21" and data "00:00"
2022.10.26 16:44:31.461 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x8002" with data type "0x30" and data "02"
2022.10.26 16:44:31.464 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:31.468 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x4001" with data type "0x21" and data "00:00"
2022.10.26 16:44:31.471 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x4002" with data type "0x21" and data "00:00"
2022.10.26 16:44:31.474 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x8002" with data type "0x30" and data "02"
2022.10.26 16:44:31.568 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD030" with data type "0x30" and data "00"
2022.10.26 16:44:31.575 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD000" with data type "0x20" and data "00"
2022.10.26 16:44:31.578 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD001" with data type "0x1B" and data "00:00:00:00"
2022.10.26 16:44:31.581 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD002" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.585 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD003" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.591 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD004" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.594 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD005" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.597 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD030" with data type "0x30" and data "00"
2022.10.26 16:44:31.601 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD000" with data type "0x20" and data "00"
2022.10.26 16:44:31.604 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD001" with data type "0x1B" and data "00:00:00:00"
2022.10.26 16:44:31.609 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD002" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.613 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD003" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.617 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD004" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:31.620 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0xE001" attribute "0xD005" with data type "0x23" and data "00:00:00:00"
2022.10.26 16:44:33.793 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "01"
2022.10.26 16:44:33.799 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "01"
2022.10.26 16:44:34.572 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:34.575 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:34.777 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:34.780 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:34.973 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:34.976 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
[... cut ...]
2022.10.26 16:44:35.178 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:35.374 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"
2022.10.26 16:44:35.377 (wrn) zigbee:    No property found for device "a4:c1:38:6a:08:33:da:bc" endpoint "0x01" cluster "0x0006" attribute "0x0000" with data type "0x10" and data "00"

d51x, 2022-10-26:
https://aliexpress.ru/item/1005004405691474.html

u236, 2022-10-30:
Added (partially) in https://github.com/u236/homed-service-zigbee/commit/28aeb9e8c241788670b4de33044538ac4e303d05.

u236, 2022-10-30:
Added in https://github.com/u236/homed-service-zigbee/commit/d3dc48a544cf043612d82aca37a4a979bc428263.

closed 2022-10-30 by u236