u236/homed-service-zigbee #8: Sonoff Door Sensor SNZB-04

d51x, 2022-10-26:
Device "00:12:4b:00:25:12:af:1c" joined network with address "0x1646"
2022.10.26 16:07:33.553 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" node descriptor received, manufacturer code: "0x0000"
2022.10.26 16:07:34.612 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" active endpoints received: "0x01"
2022.10.26 16:07:35.153 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" endpoint "0x01" simple descriptor received
2022.10.26 16:07:37.841 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" endpoint "0x01" IAS zone enrolled
2022.10.26 16:07:37.848 (wrn) zigbee:    Device "00:12:4b:00:25:12:af:1c" model name "DS01" unrecognized
2022.10.26 16:07:37.851 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" manufacturer name is "eWeLink" and model name is "DS01"
2022.10.26 16:07:37.854 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" interview finished
2022.10.26 16:07:49.081 (wrn) zigbee:    No property found for device "00:12:4b:00:25:12:af:1c" endpoint "0x01" cluster "0x0500" command "0x00" with payload "00:00:00:42:00:00:02"
2022.10.26 16:07:51.330 (wrn) zigbee:    No property found for device "00:12:4b:00:25:12:af:1c" endpoint "0x01" cluster "0x0500" command "0x00" with payload "01:00:00:42:00:00:02"

u236, 2022-10-29:
Added in https://github.com/u236/homed-service-zigbee/commit/75f7e5e16f9f0eac333bb6f4ea45aa67a05ff075.

d51x, 2022-10-29:
not working

2022.10.29 20:48:34.811 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" joined network with address "0xB7F1"
2022.10.29 20:48:35.507 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" node descriptor received, manufacturer code: "0x0000"
2022.10.29 20:48:36.043 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" active endpoints received: "0x01"
2022.10.29 20:48:36.577 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" endpoint "0x01" simple descriptor received
2022.10.29 20:48:51.335 (wrn) zigbee:    Device "00:12:4b:00:25:12:af:1c" interview timed out
2022.10.29 20:49:48.689 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" left network
2022.10.29 20:49:53.655 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" joined network with address "0x4731"
2022.10.29 20:49:54.358 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" node descriptor received, manufacturer code: "0x0000"
2022.10.29 20:49:54.888 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" active endpoints received: "0x01"
2022.10.29 20:49:55.427 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" endpoint "0x01" simple descriptor received
2022.10.29 20:50:08.510 (inf) zigbee:    Device "00:15:8d:00:00:b5:62:7d" rejoined network with address "0x3AA3"
2022.10.29 20:50:10.336 (wrn) zigbee:    Device "00:12:4b:00:25:12:af:1c" interview timed out
2022.10.29 20:50:34.096 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" left network
2022.10.29 20:50:38.986 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" joined network with address "0xE45A"
2022.10.29 20:50:39.697 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" node descriptor received, manufacturer code: "0x0000"
2022.10.29 20:50:40.233 (inf) zigbee:    Device "00:12:4b:00:25:12:af:1c" active endpoints received: "0x01"
2022.10.29 20:50:55.336 (wrn) zigbee:    Device "00:12:4b:00:25:12:af:1c" interview timed out

u236, 2022-11-01:
fixed in https://github.com/u236/homed-service-zigbee/commit/db1ed02b5ce411db3d937f478e01760b6de3a444