u236/homed-service-zigbee #21: Konke 2AJZ4KPBS таймаут по умолчанию

u236, 2022-11-25:
сделать таймаут по умолчанию равным 2 минутам и добавить пометку на вики о том, что этот датчик - тупой

Omirax, 2022-11-25:
Датчик не тупой, а позиционируется как аларм датчик скрытой установки. Он поэтому не мигает, когда срабатывает. Но отсутствие функции повторной отправки обнаружения движения или прекращения движения - ограничивает его область применения в системе УД

u236, 2022-11-25:
> Датчик не тупой, а позиционируется как аларм датчик скрытой установки. Он поэтому не мигает, когда срабатывает. Но отсутствие функции повторной отправки обнаружения движения или прекращения движения - ограничивает его область применения в системе УД

Хорошо, тогда я попрошу тебя написать текст предупреждения для вики, потому что я не очень понимаю как это корректно донести :)

u236, 2022-11-25:
поменял таймаут по умолчанию в https://github.com/u236/homed-service-zigbee/commit/f681d64bc90ee94ba283bb67f49cbe6e5f17e7c9

closed 2022-12-01 by u236