u236/homed-service-zigbee #22: Опция конфигурации для разрешения/запрета использования опций

u236, 2022-11-25:
нужно уточнение, что именно должна "запрещать" эта натсройка, на сколько я понимаю, имеется в виду глобальное отключение использования пользовательских опций

Omirax, 2022-11-25:
Нет. Глобальное запрещение использования не всех опций, а именно опций калибровки параметров.

u236, 2022-11-25:
получается, что нужно отфильтровать по флагу [опции](https://wiki.u236.org/page/ZigBee/Библиотека_устройств/Опции), название которых содержит "Offset"?

u236, 2022-11-25:
Реализовал в https://github.com/u236/homed-service-zigbee/commit/856a3921120f422e4a4da8b0720bda8d12ae5f9f. Для использования фичи нужно добавить в секцию конфига <code>[zigbee]</code> параметр <code>offsets</code>:
```
[zigbee]
...
offsets=false
...
```
<code>true</code> - корректировка разрешена, <code>false</code> - запрещена (корректировки не будут применяться к данным)

closed 2022-11-25 by u236

u236, 2022-11-29:
в https://github.com/u236/homed-service-zigbee/commit/7b550a82b93feaaa10a4bb3f67f90deea370bf74 параметр конфигурации перенесен в секцию <code>[device]</code>