u236/homed-service-zigbee #31: Добавить датчик присутствия WZ-M100-W

Sborkahome, 2022-12-28:
[presence.txt](https://github.com/u236/homed-service-zigbee/files/10313366/presence.txt)

u236, 2022-12-28:
Такие датчики уже поддерживаются, но у этого конкретного новый идентификатор модели, которого не было, когда я добавлял поддержку. Добавил новые иденификаторы в https://github.com/u236/homed-service-zigbee/commit/fedadaaecfc3fdb10f762c9b73d67a87bc43ffff

Чтобы не ждать нового релиза, можно вручную обновить [файл](https://github.com/u236/homed-service-zigbee/blob/fedadaaecfc3fdb10f762c9b73d67a87bc43ffff/deploy/data/usr/share/homed/zigbee.json) библиотеки, путь по умолчанию: `/usr/share/homed/zigbee.json`

Документация по этим датчикам [здесь](https://wiki.u236.org/page/ZigBee/%D0%9F%D0%BE%D0%B4%D0%B4%D0%B5%D1%80%D0%B6%D0%B8%D0%B2%D0%B0%D0%B5%D0%BC%D1%8B%D0%B5_%D1%83%D1%81%D1%82%D1%80%D0%BE%D0%B9%D1%81%D1%82%D0%B2%D0%B0/%D0%94%D0%B0%D1%82%D1%87%D0%B8%D0%BA%D0%B8_%D0%BF%D1%80%D0%B8%D1%81%D1%83%D1%82%D1%81%D1%82%D0%B2%D0%B8%D1%8F_TUYA)

closed 2022-12-28 by u236

Sborkahome, 2022-12-28:
А можно добавить такие параметр, как fading_time и target_distance?

u236, 2022-12-28:
у этого датчика "target_distance" разделен на два параметра - "distanceMin" и "distanceMaх", они реализованы, а "fading_time" я так и не понял зачем нужен, поэтому решил его не добавлять

u236, 2022-12-28:
добавил в https://github.com/u236/homed-service-zigbee/commit/37ddd90da4e6f2284cce6ad0f1dad1069f7543d8