u236/homed-service-zigbee #32: Add support LoraTap 3 gang wireless switch remote TS0043

KeNt606, 2022-12-28:
`2022.12.28 18:31:06.193 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" joined network with address "0x077e"

2022.12.28 18:31:06.197 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" interview started...
2022.12.28 18:31:07.011 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" node descriptor received, manufacturer code is "0x1002" and logical type is "end device"
2022.12.28 18:31:08.632 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" active endpoints received: "0x01, 0x02, 0x03"
2022.12.28 18:31:09.357 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" simple descriptor received
2022.12.28 18:31:09.537 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x02" simple descriptor received
2022.12.28 18:31:10.332 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x03" simple descriptor received
2022.12.28 18:31:11.147 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "42"
2022.12.28 18:31:11.150 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0004" report received with type "0x42" and data "5f:54:5a:33:30:30:30:5f:62:69:36:6c:70:73:65:77"
2022.12.28 18:31:11.154 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0005" report received with type "0x42" and data "54:53:30:30:34:33"
2022.12.28 18:31:11.158 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0007" report received with type "0x30" and data "03"
2022.12.28 18:31:11.162 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" manufacturer name is "_TZ3000_bi6lpsew" and model name is "TS0043"
2022.12.28 18:31:11.187 (wrn) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" manufacturer name "_TZ3000_bi6lpsew" and model name "TS0043" unrecognized
2022.12.28 18:31:11.191 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" properties recognized: "battery, action"
2022.12.28 18:31:11.194 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" discoveries: "battery"
2022.12.28 18:31:11.198 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x02" properties recognized: "battery, action"
2022.12.28 18:31:11.201 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x02" discoveries: "battery"
2022.12.28 18:31:11.204 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x03" properties recognized: "battery, action"
2022.12.28 18:31:11.208 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x03" discoveries: "battery"
2022.12.28 18:31:11.211 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" identified as "Unsupported Device _TZ3000_bi6lpsew/TS0043"
2022.12.28 18:31:11.214 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" interview finished successfully
2022.12.28 18:31:11.439 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "42"
2022.12.28 18:31:11.442 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report received with type "0x20" and data "1c"
2022.12.28 18:31:11.446 (wrn) zigbee:    No property found for device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report with
type "0x20" and data "1c"
2022.12.28 18:31:12.449 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "42"
2022.12.28 18:31:12.453 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report received with type "0x20" and data "1c"
2022.12.28 18:31:12.456 (wrn) zigbee:    No property found for device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report with
type "0x20" and data "1c"
2022.12.28 18:31:13.458 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "42"
2022.12.28 18:31:13.461 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report received with type "0x20" and data "1c"
2022.12.28 18:31:13.465 (wrn) zigbee:    No property found for device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffe2" report with
type "0x20" and data "1c"
2022.12.28 18:31:15.478 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0001" attribute "0x0021" report received with type "0x20" and data "c8"
2022.12.28 18:31:15.486 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0001" attribute "0x0020" report received with type "0x20" and data "1e"
2022.12.28 18:31:17.509 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffdf" report received with type "0x42" and data "41:41:3f:2b:12:a5:44:3f:2b:12:df:47:3f:2b:12:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:11:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c"
2022.12.28 18:31:17.514 (wrn) zigbee:    No property found for device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0000" attribute "0xffdf" report with
type "0x42" and data "41:41:3f:2b:12:a5:44:3f:2b:12:df:47:3f:2b:12:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:11:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c:00:00:00:00:0c"
2022.12.28 18:31:19.514 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" requested local time`

u236, 2022-12-28:
Please, add debug log messages, received when you press one of buttons once and double times, or hold it

KeNt606, 2022-12-28:
> 

2022.12.29 00:21:17.898 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "00"
2022.12.29 00:21:18.908 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "00"
2022.12.29 00:21:23.835 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "01"
2022.12.29 00:21:24.846 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "01"
2022.12.29 00:21:28.918 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "01"
2022.12.29 00:21:29.930 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "01"
2022.12.29 00:21:38.608 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "02"
2022.12.29 00:21:39.624 (inf) zigbee:    Device "5c:02:72:ff:fe:85:6b:41" endpoint "0x01" cluster "0x0006" command "0xfd" received with payload "02"

u236, 2023-01-06:
добавил в https://github.com/u236/homed-service-zigbee/releases/tag/3.1.6

closed 2023-01-06 by u236