u236/homed-service-zigbee #35: Добавить ptvo-switch

Sborkahome, 2023-01-05:
[ptvo.txt](https://github.com/u236/homed-service-zigbee/files/10354995/ptvo.txt)

Sborkahome, 2023-01-05:
[Топики_mqtt_для_включения_реле.txt](https://github.com/u236/homed-service-zigbee/files/10355013/_mqtt_._._.txt)
Это пример топиков для данного реле, которое подключалось к Z2M.

u236, 2023-01-06:
Устройства с прошивкой PTVO уникальны, нет смысла добавлять их в библиотеку. В обозримом будущем, вероятно, появится возможность автоматического определения всех функций устройств с прошивкой PTVO, тогда можно будет вернуться к этому реле.

closed 2023-01-06 by u236

Sborkahome, 2023-01-06:
https://ptvo.info/zigbee-configurable-firmware-features/uart/
а может эта ссылка поможет? Работа с UARTом

u236, 2023-01-07:
> https://ptvo.info/zigbee-configurable-firmware-features/uart/
> 
> а может эта ссылка поможет? Работа с UARTом

Вопрос не в том, чтобы понять как это работает, а в том, что это уникальное устройство, существущее, скорее всего, в единственном экземпляре. Если я буду добавлять каждую подробную свистелку на PTVO в общую библиотеку, скоро библиотека будет на 99% состоять из этих свистелок. Поэтому надо подождать, пока в PTVO появится функционал, который позволит работать с такими устройствами без добавления их в библиотеку.