u236/homed-service-zigbee #36: добавить поддержку lumi.relay.c2acn01

Sborkahome, 2023-01-10:
Tue Jan 10 18:07:12 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:07:12.743 (inf) zigbee:    Permit join enabled successfully
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.578 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" joined network with address "0x9ab5"
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.582 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" interview started...
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.706 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" node descriptor received, manufacturer code is "0x1037" and logical type is "router"
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.826 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" active endpoints received: "0x01, 0x02"
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.862 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0005" report received with type "0x42" and data "6c:75:6d:69:2e:72:65:6c:61:79:2e:63:32:61:63:6e:30:31" and transaction id 3
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.866 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "01" and transaction id 3
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.915 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0xff01" report received with type "0x42" and data "03:28:00:05:21:55:00:07:27:00:00:00:00:00:00:00:00:08:21:23:12:09:21:00:01:64:10:00:65:10:00:6e:20:00:6f:20:00:94:20:0f:95:39:48:21:c2:40:96:39:bb:b3:0f:45:97:39:f8:20:8c:3d:98:39:24:aa:1a:40:9b:21:00:00" and transaction id 4
Tue Jan 10 18:08:35 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:35.991 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" simple descriptor received
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.127 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x02" simple descriptor received
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.257 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20" and data "23" and transaction id 4
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.261 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0004" report received with type "0x42" and data "4c:55:4d:49" and transaction id 4
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.264 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0005" report received with type "0x42" and data "6c:75:6d:69:2e:72:65:6c:61:79:2e:63:32:61:63:6e:30:31" and transaction id 4
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.269 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0000" attribute "0x0007" report received with type "0x30" and data "04" and transaction id 4
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.272 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" manufacturer name is "LUMI" and model name is "lumi.relay.c2acn01"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.301 (wrn) zigbee:    Device "00:15:8d:00:03:96:68:ba" manufacturer name "LUMI" and model name "lumi.relay.c2acn01" unrecognized
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.304 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" properties recognized: "temperature, status, voltage, current, power"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.308 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" actions recognized: "status"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.311 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" discoveries: "temperature, switch, voltage, current, power"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.315 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x02" properties recognized: "status"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.318 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x02" actions recognized: "status"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.321 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x02" discoveries: "switch"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.326 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" identified as "Unsupported Device LUMI/lumi.relay.c2acn01"
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.329 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" interview finished successfully
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.900 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report received with type "0x39" and data "80:5a:0e:3d" and transaction id 6
Tue Jan 10 18:08:36 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:08:36.903 (wrn) zigbee:    No property found for device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report with type "0x39" and data "80:5a:0e:3d"
Tue Jan 10 18:09:25 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:25.262 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0006" attribute "0x0000" report received with type "0x10" and data "01" and transaction id 56
Tue Jan 10 18:09:25 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:25.267 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x0006" attribute "0xf000" report received with type "0x23" and data "00:b5:9a:03" and transaction id 56
Tue Jan 10 18:09:26 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:26.905 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report received with type "0x39" and data "49:a5:d6:41" and transaction id 58
Tue Jan 10 18:09:26 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:26.908 (wrn) zigbee:    No property found for device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report with type "0x39" and data "49:a5:d6:41"
Tue Jan 10 18:09:28 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:28.904 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report received with type "0x39" and data "9c:91:01:42" and transaction id 61
Tue Jan 10 18:09:28 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:28.908 (wrn) zigbee:    No property found for device "00:15:8d:00:03:96:68:ba" endpoint "0x01" cluster "0x000c" attribute "0x0055" report with type "0x39" and data "9c:91:01:42"
Tue Jan 10 18:09:32 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:32.908 (inf) zigbee:    Device "00:15:8d:00:03:96:68:ba" requested UTC time
Tue Jan 10 18:09:57 2023 daemon.info homed-zigbee[4580]: 2023.01.10 18:09:57.765 (inf) zigbee:    Permit join disabled successfully

u236, 2023-01-12:
добавил в https://github.com/u236/homed-service-zigbee/commit/7e51c764980720d769ea7835ef0db7f6449615bb

u236, 2023-01-12:
доработал в https://github.com/u236/homed-service-zigbee/commit/ea5ea642e517a5f7e9fb835e05ed9a83939c3db5

closed 2023-01-12 by u236