u236/homed-service-zigbee #40: Добавить поддержку выключатель Aqara E1

KeNt606, 2023-01-14:
Модель aqara e1 lumi.switch.b2lc04
Лог:
```
2023.01.14 13:42:57.350 (inf) zigbee:    Device "Aqara E1 switch" rejoined network with address "0xd3a7"
2023.01.14 13:42:57.353 (inf) zigbee:    Device "Aqara E1 switch" network address updated
2023.01.14 13:42:57.356 (inf) zigbee:    Device "Aqara E1 switch" interview started...
2023.01.14 13:42:57.916 (inf) zigbee:    Device "Aqara E1 switch" node descriptor received, manufacturer code is "0x115f" and logical type is "end de
vice"
2023.01.14 13:42:58.661 (inf) zigbee:    Device "Aqara E1 switch" active endpoints received: "0x01, 0x02, 0xf2"
2023.01.14 13:42:59.099 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" simple descriptor received
2023.01.14 13:42:59.102 (inf) zigbee:    Device "Aqara E1 switch" node descriptor received, manufacturer code is "0x115f" and logical type is "end de
vice"
2023.01.14 13:42:59.486 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" simple descriptor received
2023.01.14 13:42:59.771 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" simple descriptor received
2023.01.14 13:43:01.072 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" simple descriptor received
2023.01.14 13:43:01.348 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" simple descriptor received
2023.01.14 13:43:02.155 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0xf2" simple descriptor received
2023.01.14 13:43:02.805 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20
" and data "14" and transaction id 55
2023.01.14 13:43:02.809 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0004" report received with type "0x42
" and data "4c:55:4d:49" and transaction id 55
2023.01.14 13:43:02.813 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0005" report received with type "0x42
" and data "6c:75:6d:69:2e:73:77:69:74:63:68:2e:62:32:6c:63:30:34" and transaction id 55
2023.01.14 13:43:02.817 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0007" report received with type "0x30
" and data "01" and transaction id 55
2023.01.14 13:43:02.821 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0xf2" simple descriptor received
2023.01.14 13:43:02.825 (inf) zigbee:    Device "Aqara E1 switch" manufacturer name is "LUMI" and model name is "lumi.switch.b2lc04"
2023.01.14 13:43:02.857 (wrn) zigbee:    Device "Aqara E1 switch" manufacturer name "LUMI" and model name "lumi.switch.b2lc04" properties not found i
n library
2023.01.14 13:43:02.860 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" properties recognized: "temperature, status"
2023.01.14 13:43:02.864 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" actions recognized: "status"
2023.01.14 13:43:02.867 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" discoveries: "temperature, switch"
2023.01.14 13:43:02.871 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" properties recognized: "status"
2023.01.14 13:43:02.874 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" actions recognized: "status"
2023.01.14 13:43:02.877 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x02" discoveries: "switch"
2023.01.14 13:43:02.882 (inf) zigbee:    Device "Aqara E1 switch" identified as "LUMI/lumi.switch.b2lc04"
2023.01.14 13:43:02.885 (inf) zigbee:    Device "Aqara E1 switch" interview finished successfully
2023.01.14 13:43:03.290 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0001" report received with type "0x20
" and data "14" and transaction id 55
2023.01.14 13:43:03.294 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0004" report received with type "0x42
" and data "4c:55:4d:49" and transaction id 55
2023.01.14 13:43:03.298 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0005" report received with type "0x42
" and data "6c:75:6d:69:2e:73:77:69:74:63:68:2e:62:32:6c:63:30:34" and transaction id 55
2023.01.14 13:43:03.302 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0000" attribute "0x0007" report received with type "0x30
" and data "01" and transaction id 55
2023.01.14 13:43:11.873 (inf) zigbee:    Device "Aqara E1 switch" endpoint "0x01" cluster "0x0019" command "0x01" received with payload "00:5f:11:08:
1a:14:00:00:00" and transaction id 152
```

u236, 2023-01-15:
Уточни пожалуйста IEEE-адрес выключателя

KeNt606, 2023-01-15:
> 

[image]

u236, 2023-01-16:
добавил в https://github.com/u236/homed-service-zigbee/commit/b548502c8f65490f9beac35b49d8cbd8dee12e9e

closed 2023-01-16 by u236