u236/homed-service-zigbee #50: no support TS0601

dixanl, 2023-05-20:
Good day! could you please add support for tyua ts0601 thermal head

dixanl, 2023-05-24:
<html>
<body>

Description | _TZE200_ckud7u2l/TS0601
-- | --
Logical type | end device
Power source |  
IEEE address | 60:a4:23:ff:fe:ab:48:c9
Network address | 0x8ad3
Manufacturer code | 0x1002
Manufacturer name | _TZE200_ckud7u2l
Model name | TS0601
Version | 87
Interview finished |  
Supported |  

<div class="exposes title" style="border: 0px; box-sizing: border-box; margin: 10px 4px; padding: 0px; align-items: center; display: flex; font-size: 1.25em; font-weight: 700; gap: 10px; justify-content: space-between;"><span class="name" style="border: 0px; box-sizing: border-box; margin: 0px; padding: 0px;">Exposes</span></div><table class="exposes" style="border: 0px; box-sizing: border-box; margin: 10px 0px 20px; padding: 0px; border-collapse: collapse; border-spacing: 0px; width: 910px;"><tbody style="border: 0px; box-sizing: border-box; margin: 0px; padding: 0px;"><tr data-name="linkQuality" style="border: 0px; box-sizing: border-box; margin: 0px; padding: 0px;"><td style="border: none; box-sizing: border-box; margin: 0px; padding-top: 5px; padding-right: 5px; padding-bottom: 5px; padding-left: 5px !important; white-space: nowrap; min-width: 150px; width: 227.5px; color: rgb(68, 68, 68); font-family: &quot;Open Sans&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Link quality</td><td class="value" style="border: none; box-sizing: border-box; margin: 0px; padding: 5px 20px 5px 5px; white-space: nowrap; min-width: 40px; width: 49.025px; color: rgb(68, 68, 68); font-family: &quot;Open Sans&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">144</td><br class="Apple-interchange-newline">
</body>
</html>

u236, 2023-05-24:
я помню про эту задачу, она стоит в очереди, нужно немного подождать :)

dixanl, 2023-05-24:
Спасибо! Подождать не проблема. 
Просто увидел поддержку решил проверить

u236, 2023-05-25:
добаил **частичную** поддержку в https://github.com/u236/homed-service-zigbee/commit/3e8c3c3e17554ddff7243db38438717c0e23cb8d, пока не поддерживается расписание

потестировать можно, взяв бинарник под нужную архитекутру здесь: http://sandbox.u236.org/homed-zigbee/ (нужно не забыть сделать бинарник исполняемым после замены файла)

и обновив файл библиотеки tuya.js, взяв актуальный отсюда: https://github.com/u236/homed-service-zigbee/blob/master/deploy/data/usr/share/homed-zigbee/tuya.json

просьба отписаться в чате после тестов, поддержку расписания добавлю позже

dixanl, 2023-05-25:
все поставил режим толко обогрев (Автоматический/выключено этих режимов нет)
также нет и других функций #
[image]

u236, 2023-05-25:
В HA пробрасываются только **базовые** функции термостататов (установка температуры и выбор пресета), остальными функциями можно управлять через веб-интерфейс HZB, который тоже еще нужно доработать под это устройство.

Для этой модели термостата я сделал все, что мог сделать, не имея устройства на руках, к сожалению. Довольно сложно разбираться с такой сложной штукой "по фотографии". Если найдется добрый самаритянин, готовый прислать мне устройство, я с радостью добавлю недостающие функции.

Управление расписанием я добавил, оно будет доступно в следующем релизе HZB.

closed 2023-05-25 by u236