u236/homed-service-zigbee #52: eWeLink Switch ZR-02 с прошивкой PTVO

DGMiron, 2023-05-23:
Прошивка https://t.me/zigbeeptvo/13476
Судя по конвертору для з2м: рэле - свич, кнопка - экшн, есть температура чипа.
Автораспознаётся только свич. Своими силами удалось добавить температуру чипа (не знаю, - зачем она нужна, но всё ж..)
Не осилил экшн кнопки. 
Сделал таким образом:
{
  "ptvo.switch-zr02":
  [
   {
     "description":    "my_ptvo.switch",
      "modelNames":     ["ptvo.switch-zr02"],
      "properties":     ["status"],
      "actions":        ["status"],
      "exposes":        ["switch"],
      "endpointId":     2
   },
   {
     "modelNames":     ["ptvo.switch-zr02"],
      "properties":     ["ptvoTemperature"],
      "exposes":        ["temperature"],
      "endpointId":     3
   }
   ]
}
Прошу помощи, подсказки в добавлении кнопки.

u236, 2023-05-24:
добавил новое свойство "ptvoButtonAction" в https://github.com/u236/homed-service-zigbee/commit/cdef67cc3fdc4a9127684bcf667e119c8f520e11, потестировать можно, взяв бинарник под нужную архитекутру здесь: http://sandbox.u236.org/homed-zigbee/ (нужно не забыть сделать бинарник исполняемым после замены файла)

DGMiron, 2023-05-24:
Изменил либу, проверил - работает!!! Спасибо!
Содержание библиотеки: https://pastebin.com/ETrwVjtK
[image]
[image]
[image]

u236, 2023-05-25:
https://github.com/u236/homed-service-zigbee/commit/ecf6791c3c93f18467db081510141f63e80f8b96

closed 2023-05-25 by u236