u236/homed-service-zigbee #57: TUYA TS0601 Neo Siren добавить в дисковери homeassistant

d51x, 2023-06-09:


u236, 2023-06-12:
https://github.com/u236/homed-service-zigbee/commit/7e87b76baca7fc810f35217ce39b5e7397cf9795

d51x, 2023-06-14:
в homeassistant появилась только батарейка, управление сиреной отсутствует

Вот как z2m публикует данные в топик discovery

`MQTT publish: topic 'homeassistant/switch/0xa4c1388ec5567447/switch_alarm/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/SirenNeo/set/alarm","device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"name":"SirenNeo alarm","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_switch_alarm_zigbee2mqtt","value_template":"{% if value_json.alarm %} true {% else %} false {% endif %}"}'`

` MQTT publish: topic 'homeassistant/sensor/0xa4c1388ec5567447/melody/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:music-note","name":"SirenNeo melody","state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_melody_zigbee2mqtt","value_template":"{{ value_json.melody }}"}'`

`MQTT publish: topic 'homeassistant/select/0xa4c1388ec5567447/melody/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/SirenNeo/set/melody","device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"entity_category":"config","icon":"mdi:music-note","name":"SirenNeo melody","options":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"],"state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_melody_zigbee2mqtt","value_template":"{{ value_json.melody }}"}'`

`MQTT publish: topic 'homeassistant/sensor/0xa4c1388ec5567447/duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:timer","name":"SirenNeo duration","state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_duration_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.duration }}"}'`

`MQTT publish: topic 'homeassistant/number/0xa4c1388ec5567447/duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/SirenNeo/set/duration","device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"entity_category":"config","icon":"mdi:timer","max":1800,"min":0,"name":"SirenNeo duration","state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_duration_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.duration }}"}'`

`MQTT publish: topic 'homeassistant/select/0xa4c1388ec5567447/volume/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/SirenNeo/set/volume","device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"entity_category":"config","icon":"mdi: volume-high","name":"SirenNeo volume","options":["low","medium","high"],"state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_volume_zigbee2mqtt","value_template":"{{ value_json.volume }}"}'`

`MQTT publish: topic 'homeassistant/select/0xa4c1388ec5567447/volume/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/SirenNeo/set/volume","device":{"identifiers":["zigbee2mqtt_0xa4c1388ec5567447"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"SirenNeo"},"entity_category":"config","icon":"mdi: volume-high","name":"SirenNeo volume","options":["low","medium","high"],"state_topic":"zigbee2mqtt/SirenNeo","unique_id":"0xa4c1388ec5567447_volume_zigbee2mqtt","value_template":"{{ value_json.volume }}"}'`

d51x, 2023-06-14:
[image]

u236, 2023-06-15:
https://github.com/u236/homed-service-zigbee/commit/a4ccfa24f54c21b01651ebae1de62986e3a6ae49

closed 2023-06-15 by u236