u236/homed-service-zigbee #66: добавить "Feibit Inc co." and model name is "FB56-ZCW08KU1.1"

Sborkahome, 2023-07-21:
Добрый день!
добавьте, пжт, новое устройство в библиотеку
 Device "00:12:4b:00:18:34:bd:4e" joined network with address "0x38f2"
Fri Jul 21 12:09:39 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:39.557 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" interview started...
Fri Jul 21 12:09:39 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:39.659 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" node descriptor received, manufacturer code is "0x0000" and logical type is "router"
Fri Jul 21 12:09:39 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:39.756 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" active endpoints received: "0x0b, 0x0d"
Fri Jul 21 12:09:39 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:39.856 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0b" simple descriptor received
Fri Jul 21 12:09:39 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:39.955 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0d" simple descriptor received
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.171 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0b" color capabilities: "0x000f"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.174 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" manufacturer name is "Feibit Inc co." and model name is "FB56-ZCW08KU1.1"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.379 (wrn) zigbee:    Device "00:12:4b:00:18:34:bd:4e" manufacturer name "Feibit Inc co." and model name "FB56-ZCW08KU1.1" not found in library
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.382 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0b" properties recognized: "status, level, color"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.385 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0b" actions recognized: "status, level, color"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.388 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" endpoint "0x0b" exposes: "light"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.390 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" firmware is "1.0.2"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.393 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" identified as "Feibit Inc co./FB56-ZCW08KU1.1"
Fri Jul 21 12:09:40 2023 daemon.info homed-zigbee[26101]: 2023.07.21 12:09:40.396 (inf) zigbee:    Device "00:12:4b:00:18:34:bd:4e" interview finished successfully

u236, 2023-07-21:
https://github.com/u236/homed-service-zigbee/commit/d4b3c61f5edace89ca97d7bcfb1d1436f6f5514e

closed 2023-07-21 by u236

Sborkahome, 2023-07-21:
вот так работает
{
  "Feibit Inc co.":
  [
    {
      "description":    "Feibit Inc co./FB56-ZCW08KU1.1",
      "modelNames":     ["FB56-ZCW08KU1.1"],
      "properties":     ["status", "powerOnStatus", "level", "colorXY", "colorTemperature"],
      "actions":        ["status", "powerOnStatus", "level", "colorXY", "colorTemperature"],
      "exposes":        ["light", "powerOnStatus"],
      "options":        {"light": ["level", "color", "colorTemperature"], "powerOnStatus": {"enum": ["on", "off", "toggle", "previous"]}},
      "endpointId":     11
    }
  ]
}