u236/homed-service-zigbee #107: прошу добавить датчик присутствия "_TZE204_kyhbrfyl", model name is "TS0601"

Sborkahome, 2024-01-07:
Sun Jan  7 12:11:24 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:11:24.631 (inf) zigbee:    Permit join enabled successfully
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.401 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" joined network with address "0x30db"
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.405 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" interview started...
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.698 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" node descriptor received, manufacturer code is "0x1141" and logical type is "router"
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.780 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" active endpoints received: "0x01, 0xf2"
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.870 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" endpoint "0x01" simple descriptor received
Sun Jan  7 12:12:04 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:04.961 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" endpoint "0xf2" simple descriptor received
Sun Jan  7 12:12:05 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:05.055 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" manufacturer name is "_TZE204_kyhbrfyl" and model name is "TS0601"
Sun Jan  7 12:12:05 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:05.183 (wrn) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" manufacturer name "_TZE204_kyhbrfyl" and model name "TS0601" not found in library
Sun Jan  7 12:12:05 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:05.188 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" identified as "_TZE204_kyhbrfyl/TS0601"
Sun Jan  7 12:12:05 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:05.191 (inf) zigbee:    Device "a4:c1:38:be:c5:d5:53:9b" interview finished successfully
Sun Jan  7 12:12:43 2024 daemon.info homed-zigbee[2873]: 2024.01.07 12:12:43.337 (inf) zigbee:    Permit join disabled successfully

u236, 2024-01-09:
https://github.com/u236/homed-service-zigbee/commit/95b744cc111d6a4bf7fd02b9acc7828655858017

closed 2024-01-09 by u236

Sborkahome, 2024-01-11:
немного изменил:
    {
      "description":    "TUYA TS0601 Human Presence Sensor (Type E)",
      "modelNames":     ["_TZE204_kyhbrfyl"],
      "properties":     ["tuyaDataPoints"],
      "actions":        ["tuyaDataPoints"],
      "exposes":        ["occupancy", "motionStatus", "targetDistance", "sensitivity", "motionSensitivity", "distanceMax", "fadingTime"],
      "options":        {
                          "tuyaDataPoints":
                          {
                            "1":    [{"name": "occupancy", "type": "bool"}],
                            "11":   [{"name": "motionStatus", "type": "enum"}],
                            "12":   [{"name": "fadingTime", "type": "value", "action": true}],
                            "13":   [{"name": "distanceMax", "type": "value", "divider": 100, "action": true}],
                            "15":   [{"name": "sensitivity", "type": "value", "action": true}],
                            "16":   [{"name": "motionSensitivity", "type": "value", "action": true}],
                            "19":   [{"name": "targetDistance", "type": "value", "divider": 100}]
                          },

                          "motionStatus": {"enum": ["none", "small", "large"]},
                          "fadingTime": {"min": 0, "max": 300, "step": 10},
                          "distanceMax": {"min": 1.5, "max": 6, "step": 0.1},
                          "sensitivity": {"min": 0, "max": 7},
                          "motionSensitivity": {"min": 1, "max": 7}
                        }
    },

u236, 2024-01-11:
да, спасибо, обновил - https://github.com/u236/homed-service-zigbee/commit/a575bf8de726f96596f49c5d74d54de39bc08470