u236/homed-service-zigbee #109: Поддержка флага MQTT Retain

DrBorland, 2024-01-09:
Добрый день!

Большое спасибо за замечательную платформу домашней автоматизации.

Возможно ли добавление в конфигурацию  флага MQTT Retain?

Настроена интеграция с MQTT-сервером Home Assistant через MQTT Bridge. При перезагрузке сервера Home Assistant у всех передаваемых из HOMEd объектов значится статус 'unknown', что особенно неудобно для датчиков движения, от которых обновления статуса можно ждать очень долго.

u236, 2024-01-09:
Привет, спасибо за приятные слова! Ретейнить сообщения для отдельных устройств можно при помощи файла опций, путь к которому указывается в файле конфигурации `homed-zigbee.conf`:
```
options=/opt/homed-zigbee/options.json
```
Например:
```json
{
  "11:22:33:44:55:66:77:88": {"retain": true},
  "88:77:66:55:44:33:22:11": {"retain": true, "someOption": 42},
}
```
Но ваша проблема, судя по всему, не в этом. Служба сама отлично умеет отслеживать перезапуски ХА. Проверьте, настроена ли у вас в интеграции MQTT опция **_birth message_**, и, если настроена, проверьте, что топик совпадает с параметром `homeassistant > status` в файле конфигурации `homed-zigbee.conf`. Если все настроено верно, homed-zigbee будет сам публиковать **все** известные данные устройств, когда ХА сообщает о том, что запустился.  Заодно я бы проверил, прокидывается ли топик **_birth message_** через ваш mqtt-мост. Кроме того, данные устройств публикуются и при перезапуске самого homed-zigbee.

[image]

Приходите к нам в телеграм-чат - https://t.me/homed_zigbee, там проще будет разобраться с проблемой.

DrBorland, 2024-01-10:
Спасибо за исчерпывающий ответ!

Действительно при перезапуске homed-zigbee статусы в HA подтянулись.
**birth message** в HA включен, топики совпадают (всё везде настроено по дефолту, я только пароли указал).

Видимо какая-то локальная проблема. Напишу в чат, если не смогу разобраться.

u236, 2024-01-11:
> Видимо какая-то локальная проблема
Я бы таки проверил, что топик homeassistant/status долетает до брокера, к которому подключен homed-zigbee

u236, 2024-01-17:
> Видимо какая-то локальная проблема. Напишу в чат, если не смогу разобраться.

Удалось решить проблему?

closed 2024-01-17 by u236