u236/homed-service-zigbee #118: Датчики температуры "TUYA TS0601 Temperature and Humidity Sensor (Type B)"

yser1, 2024-01-18:
День добрый.

Перебираюсь с zigbee2mqtt на homed-zigbee, столкнулся со следующей проблемой:
есть 2 датчика температуры "TUYA, TS0601, _TZE200_yjjdcqsq", "TUYA, TS0601, _TZE204_upagmta9" определяются как "TUYA TS0601 Temperature and Humidity Sensor (Type B)", после подключения данные с их не появляются (кроме linkQuality). Вчера вечером подключил их и до утра никаких данных не появилось. Сегодня переустановил homed-zigbee подключил датчики - тоже самое, данных нету.  

логи [homed.log](https://github.com/u236/homed-service-zigbee/files/13974433/homed.log)
датчик1 [a4c138efa3a6a13c.json](https://github.com/u236/homed-service-zigbee/files/13974434/a4c138efa3a6a13c.json)
датчик2 [a4c1383c44845086.json](https://github.com/u236/homed-service-zigbee/files/13974435/a4c1383c44845086.json)

PS:
В homed-zigbee с датчика "Sonoff Temperature and Humidity Sensor SNZB-02" сразу появляются все данные.
В zigbee2mqtt данные  с датчиков TUYA появляются сразу.

u236, 2024-01-18:
немного опоздали к релизу, думаю дело в magic-запросе, я [добавил](https://github.com/u236/homed-service-zigbee/commit/61c1fc80fbce866dd65326cb2f3f86072071c91a) его в библиотеку, попробуйте обновть файл `/usr/share/homed-zigbee/tuya.json` заменив эго на [этот](https://raw.githubusercontent.com/u236/homed-service-zigbee/61c1fc80fbce866dd65326cb2f3f86072071c91a/deploy/data/usr/share/homed-zigbee/tuya.json), а потом переджойнить датчики, если все заработает или не заработает, дайте знать

yser1, 2024-01-18:
удалил датчики, заменил файл tuya.json, добавил датчики, результат не изменился :(
в логе вижу появилась новая строка "magic request finished successfully" , а в остальном все то же самое, почему то датчики все время переподключаются к сети, а не переходят в режим измерения. 

лог [144640_homed.log](https://github.com/u236/homed-service-zigbee/files/13975626/144640_homed.log)

PS:
если нужен более подробный лог, скажи какие параметры в настройках включить для этого.

u236, 2024-01-18:
> PS: если нужен более подробный лог, скажи какие параметры в настройках включить для этого.

лог не нужен, я добавил еще немного "магии", нужно снова обновить tuya.json, а так же обновить бинарник `/usr/bin/homed-zigbee`, взяв тестовый для своей архитектуры вот [здесь](http://sandbox.u236.org/homed-zigbee/), и не забыть сделать его исполняемым:
```
# chmod +x /usr/bin/homed-zigbee
```
после перезапуска службы датчик нужно переджойнить, надеюсь в этот раз заработает

yser1, 2024-01-18:
все заработало! спасибо!

u236, 2024-01-18:
> все заработало! спасибо!

отлично

closed 2024-01-18 by u236