u236/homed-service-zigbee #120: Таймер на permit join

Blacksli, 2024-01-18:
Такая функция есть и  zha и  z2m 
В интерфейсе все по чему видно включен или нет джойн - маленькая иконка, которая меняет цвет.
Не понятно сходу. Приходится следить за ней всегда. 

Когда заводишь устройства в любой шлюз как правило заводишь 1-2 устройства, это после первичной массовой настройки. купил устройство - завел, смотришь что умеет и т.п. 

И забываешь о пермите (да и зачем когда это можно автоматом)
И потом позалетают свои устройства с других шлюзов, соседские устройства у кого так же есть зигби и т.д. и .т.п.

очень не хватает.

Blacksli, 2024-01-18:
автор не рассматривает новые функции.

u236, 2024-01-18:
уточнение - автор рассматривает новые функции, но оставляет за собой права решать, нужны ли они

closed 2024-01-19 by u236