u236/homed-service-zigbee #130: Поддержка TS0011 ( _TZ3000_hafsquare )

seregahere, 2024-02-24:
Здравствуйте, прошу добавить поддержку безнулевого реле: 
_TZ3000_hafsqare/TS0011

[image]

Если это поможет, похожее Вы уже недавно добавляли https://github.com/u236/homed-service-zigbee/issues/121

В zigbee2mqtt определяется как https://www.zigbee2mqtt.io/devices/TS0011.html (на скрине такой же выключатель, из одной партии)
[image]

Спасибо!

seregahere, 2024-02-24:
Только что попробовал проапдейтить /usr/share/homed-zigbee/tuya.json
После рестарта homed-zigbee реле стало поддерживаемым:
[image]

Если Вы не против, могу сделать PR.

u236, 2024-02-24:
> Если Вы не против, могу сделать PR.

Да, конечно, буду только рад :)

closed 2024-02-25 by u236