u236/homed-service-zigbee #136: Добавить поддержку лампы CK-BL702-AL-01

RavenHuginn, 2024-03-11:
Прошу рассмотреть возможность добавить поддержку лампы.

[image]

https://www.zigbee2mqtt.io/devices/CK-BL702-AL-01.html

u236, 2024-03-17:
почему закрыли ишью? лампа же еще не поддерживается :)

RavenHuginn, 2024-03-17:
Нашёл решение, добавление информации по устройству в файл sonoff.json.

`,
    {
      "description":    "eWeLink/CK-BL702-AL-01(7009_Z102LG03-1)",
      "modelNames":     ["CK-BL702-AL-01(7009_Z102LG03-1)"],
      "properties":     ["status", "level", "colorXY", "colorTemperature"],
      "actions":        ["status", "level", "colorXY", "colorTemperature"],
      "exposes":        ["light"],
      "options":        {"light": ["level", "color", "colorTemperature"], "colorTemperature": {"min": 142, "max": 500}, "powerSource": 1}
    }`

Не совсем знаю как это всё правильно оформлять

u236, 2024-03-17:
я сделаю, пускай пока ишью повисит открытым :)

u236, 2024-03-18:
https://github.com/u236/homed-service-zigbee/commit/1ed99075f0e1a215e6c4a5b619df63ea2856d1ce

closed 2024-03-18 by u236