u236/homed-service-zigbee #139: Add support for TUYA TS0601 Human Presence Sensor (_TZE204_ijxvkhd0)

n8065, 2024-03-17:
Есть туёвый датчик присутствия, работающий в диапазоне 24ГГц ([блок свойств из интерфейса](https://github.com/u236/homed-service-zigbee/files/14626272/props.json))

HOMEd-zigbee v3.7.0

Заводился путём адаптации описания близкого датчика. В итоге получен [такой блок конфигурации](https://github.com/u236/homed-service-zigbee/files/14626281/_TZE204_ijxvkhd0-config.json).

В процессе была выявлена проблема с изменением параметров "Sensitivity" и "Motion sensitivity". Визуально это проявляется таким образом, что при задании нового значения параметра, слайдеру, за него отвечающему, "сносит крышу".

В ходе изучения проблемы выяснилось, что при записи эти параметров в устройство, они должны записываться "как есть", без применения делителя. А вот в рапортах девайс шлёт их значения, умноженные на 10. Всё это многократно проверялось сниффером. Как в связке с HOMEd, так и с туёвым шлюзом.

Не нашёл способа решить проблему с помощью имеющихся инструментов, поэтому в качестве quick-and-dirty fix решил добавить атрибуту divider дополнительные "грани" - атрибуты с именами propertyDividev & actionDivider. 

Думаю, есть более изящный способ решения проблемы, но в качестве подтверждения верности теории этот вариант оказался вполне рабочим. Сейчас блок конфигурации для устройства [выглядит так](https://github.com/u236/homed-service-zigbee/files/14626388/_TZE204_ijxvkhd0-config-new.json).

Прошу рассмотреть возможность поддержки таких вот хитровыделанных девайсов.

Свои изменения в коде могу подготовить в виде PR (хотя и не думаю, что это требуется).
Если нужно, готов провести дополнительные эксперименты с оборудованием.

u236, 2024-03-17:
кидай пуллреквест, посмотю :)
я правильно понял, что в https://github.com/u236/homed-service-zigbee/pull/138 другой датчик?

n8065, 2024-03-17:
Да, в #138 другой.

u236, 2024-03-17:
смерджил PR, жду новый с датчиком :)

n8065, 2024-03-18:
Спасибо.

Вечером соберу бинарник, проверю работу с учётом изменений в коде и на этой основе подготовлю PR с настройками датчика.

Ну и это issue закрою, если всё ок.

PR #144 (2024-03-18): _TZE204_ijxvkhd0 settings adjustment
Closes #139
closed 2024-03-18 by u236