u236/homed-service-zigbee #142: сенсор Perenio Motion Sensor PECMS01 после последнего обновления не работает параметр  Occupancy

cgtut, 2024-03-17:
их несколько штук и все сразу перестали работать параметр Occupancy всё время false, 
[image]
[image]
[image]

u236, 2024-03-17:
в последний раз класс, работающий со свойством occupancy для этих датчиков, менялся почти [2 месяца](https://github.com/u236/homed-service-zigbee/commit/73c8e30075ed157b694560c21c38ca1b85ff9245) назад, сомневаюсь, что проьблема связана с обновленем, я бы попробовал переджойнить один из датчиков, а заодно прислать сюда лог интервью

cgtut, 2024-03-17:
пробовал переджойнить , с 3 раза заработал 1 из 3х, мучался  , потом переустановил homed на переньке всё заработало

closed 2024-03-17 by cgtut