u236/homed-service-zigbee #147: Zigbee датчик температуры и влажности eWeLink/SNZB-02P - Not Supported, но работает

yaadyaad, 2024-03-19:
[image]

```json
{
  "active": true,
  "cloud": false,
  "description": "eWeLink/SNZB-02P",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 770,
      "endpointId": 1,
      "inClusters": [
        0,
        1,
        3,
        32,
        1026,
        1029,
        64599,
        64529
      ],
      "outClusters": [
        25
      ],
      "profileId": 260
    }
  ],
  "firmware": "2.1.0",
  "ieeeAddress": "34:10:f4:ff:fe:fa:d0:48",
  "interviewFinished": true,
  "lastSeen": 1710855716,
  "linkQuality": 21,
  "logicalType": 2,
  "manufacturerCode": 4742,
  "manufacturerName": "eWeLink",
  "modelName": "SNZB-02P",
  "name": "BalconyTempHumidity",
  "networkAddress": 11082,
  "powerSource": 3,
  "supported": false
}
```

yaadyaad, 2024-03-19:
https://www.zigbee2mqtt.io/devices/SNZB-02P.html

yaadyaad, 2024-03-19:
Добавил в sonoff.json по аналогии поддержку устройства. Строка 29 и далее. Пользовался описанием устройства тут https://www.zigbee2mqtt.io/devices/SNZB-02P.html . Почему-то пропала батарея.

```json
    {
      "description":    "Sonoff Temperature and Humidity Sensor SNZB-02P",
      "modelNames":     ["SNZB-02P"],
      "properties":     ["batteryPercentage", "temperature", "humidity", "voltage"],
      "bindings":       ["battery", "temperature", "humidity"],
      "reportings":     ["batteryPercentage", "temperature", "humidity", "voltage"],
      "exposes":        ["battery", "batteryPercentage" ,"temperature", "humidity", "voltage", "temperatureCalibration", "temperaturePrecision", "humidityCalibration", "humidityPrecision"],
      "options": {
        "temperatureCalibration": {"min": -30, "max": 30},
        "temperaturePrecision": {"min": 0, "max": 3},
        "humidityCalibration": {"min": -30, "max": 30},
        "humidityPrecision": {"min": 0, "max": 3}
      }
    },
```
Поэтому пока пул-реквест не делаю. 

[image]

yaadyaad, 2024-03-19:
[image]

yaadyaad, 2024-03-19:
[image]

yaadyaad, 2024-03-19:
[image]

yaadyaad, 2024-03-19:
Battery подтянулась, а остального похоже нет в девайсе, хотя я брал название свойств из m2z
[image]

u236, 2024-03-20:
очень много лишнего понаписал :) то что ты принял за настройки корректировок показателей на самом деле индивидуальные настройки репортига (непонятно зачем), а так этот датчик, судя по всему вообще ничем не отличается от своего старшего брата `SNZB-02`, так что пусть будет вот так: https://github.com/u236/homed-service-zigbee/commit/015148c46503ae19654691d94358511bb02fec43

closed 2024-03-20 by u236