u236/homed-service-zigbee #177: Добавить поддержку "_TZE204_cjbofhxw"

yser1, 2024-06-03:
День добрый, прошу добавить поддержку бесконтактного энергомониторинга на одну фазу 
[image]

{
  "active": true,
  "cloud": false,
  "description": "_TZE204_cjbofhxw/TS0601",
  "discovery": false,
  "endpoints": [
    {
      "deviceId": 81,
      "endpointId": 1,
      "inClusters": [
        0,
        4,
        5,
        61184
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    }
  ],
  "ieeeAddress": "38:39:8f:ff:fe:8d:e8:f8",
  "interviewFinished": true,
  "lastSeen": 1717396971,
  "linkQuality": 112,
  "logicalType": 1,
  "manufacturerCode": 4098,
  "manufacturerName": "_TZE204_cjbofhxw",
  "modelName": "TS0601",
  "neighbors": [
    {
      "linkQuality": 154,
      "networkAddress": 0
    },
    {
      "linkQuality": 82,
      "networkAddress": 26621
    }
  ],
  "networkAddress": 38687,
  "powerSource": 1,
  "supported": false,
  "version": 70,
  "name": "38:39:8f:ff:fe:8d:e8:f8"
}

yser1, 2024-06-06:
Спасибо за быстрое добавление устройства, но к сожалению данных никаких нет (см. скриншот), сразу писать не стал т.к. устройство на даче, а вот вчера вечером был на даче и думал что удалив и заново добавив устройство данные появятся - не помогло.  Подскажите, может какие логи нужно собрать и прислать?

[image]
?

yser1, 2024-06-06:
нашел в настройка параметры для лога и включил запись логов zigbee

[homed.log](https://github.com/user-attachments/files/15613215/homed.log)

u236, 2024-06-06:
> нашел в настройка параметры для лога и включил запись логов zigbee
> 
> 
> 
> [homed.log](https://github.com/user-attachments/files/15613215/homed.log)
> 
> 

Завтра посмотрю или сегодня ближе к ночи. Устройство удаляли перед повторной привязкой? 

Пока предлагаю попробовать добавить в описание устройства опцию `"tuyaMagic": true`, как у некоторых других устройств в  tuya.json, а потом опять перепривязать устройство с удадением.

yser1, 2024-06-06:
Да, перед привязкой удалял.

Ок, попробою с добавлением опции, но это уже будет в субботу.

u236, 2024-06-07:
https://github.com/u236/homed-service-zigbee/commit/2110697c8237299e02f1efe9d95c80503a7741b4

u236, 2024-06-09:
> но это уже будет только в субботу.

привет, удалось проверить?

yser1, 2024-06-09:
Привет, вот только вернулся. 
Добавлением опции "tuyaMagic": true не помогло. 

[homed.log](https://github.com/user-attachments/files/15753696/homed.log)

[image]

u236, 2024-06-09:
> Привет, вот только вернулся. 
> 
> Добавлением опции "tuyaMagic": true не помогло. 
> 
> 
> 
> [homed.log](https://github.com/user-attachments/files/15753696/homed.log)
> 
> 
> 
> 
> 
> 

Вместо лога загружается скриншот. После добавления опциии устройство переподключили? С удалением?

yser1, 2024-06-09:
Да, устройство переподключал с удалением.

[homed.log](https://github.com/user-attachments/files/15753880/homed.log)

u236, 2024-06-09:
> Да, устройство переподключал с удалением.
> 
> 
> 
> [homed.log](https://github.com/user-attachments/files/15753880/homed.log)
> 
> 

В таком случае мне жаль, без устройства на руках я больше ничем не могу помочь :(

u236, 2024-06-10:
нашел вероятный источник проблемы - устройство отправляет запросы на чтение некоего "состояния подключения шлюза к интернету" и ждет ответа, а homed-zigbee, по очевидной причине, не отвечает

добавил формирование ответа в https://github.com/u236/homed-service-zigbee/commit/73d96bb29342deb6812fd88312596869a5713382

просьба взять тестоый бинарник из [песочницы](http://sandbox.u236.org/homed-zigbee/) и проверить на нем, заработало ли устройство (как обычно, с удалением и переподключением)

closed 2024-06-10 by u236