u236/homed-service-zigbee #179: Добавить поддержку JetHome/WS7

vadimnpechorin, 2024-06-10:
JetHome/WS7
[image]
**Устройство:** https://jethome.ru/ws7
**Конвертер:** https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/jethome.ts

**Device info:** [JetHome_WS7.json](https://github.com/user-attachments/files/15760069/JetHome_WS7.json)

u236, 2024-06-10:
прошу собрать лог нажатий на клавиши (с вкюченной опцией debug > zigbee = true в конфиге), жалательно с комментариями (нажал то, нажал это)

vadimnpechorin, 2024-06-10:
Лог нажатий ключей(контактов):
[debug_jethome+_ws7.txt](https://github.com/user-attachments/files/15774661/debug_jethome%2B_ws7.txt)

u236, 2024-06-10:
https://github.com/u236/homed-service-zigbee/commit/061f2a7024bf3ed0d4910035ed8782926046bbc5

closed 2024-06-10 by u236