u236/homed-service-zigbee #197: Прошу добавить в поддержку датчик GIRIER ZD01

DGMiron, 2024-07-29:
Ссылка на анонс в Zigbee-пульс  https://t.me/zigbeat/2800?single
Инфа из вэба:
{
  "active": true,
  "cloud": true,
  "description": "_TZE200_4cqhd2ha/TS0601",
  "discovery": true,
  "endpoints": [
    {
      "colorCapabilities": 65535,
      "deviceId": 81,
      "endpointId": 1,
      "inClusters": [
        4,
        5,
        61184,
        0
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    }
  ],
  "ieeeAddress": "a4:c1:38:b1:dc:e9:d9:7a",
  "interviewFinished": true,
  "lastSeen": 1722277141,
  "linkQuality": 80,
  "logicalType": 2,
  "manufacturerCode": 4417,
  "manufacturerName": "_TZE200_4cqhd2ha",
  "modelName": "TS0601",
  "networkAddress": 60554,
  "powerSource": 3,
  "supported": false,
  "version": 66,
  "name": "a4:c1:38:b1:dc:e9:d9:7a"
}
Лог интервью в аттаче. 
[log_mts_4.txt](https://github.com/user-attachments/files/16417384/log_mts_4.txt)
[zd01_girier.txt](https://github.com/user-attachments/files/16417385/zd01_girier.txt)

u236, 2024-07-31:
никакой информации не могу найти по этому датчику, так что нужен или сам датчик для опытов, или ждать, пока кто-то добвит его в z2m/zha

u236, 2024-08-28:
прошел месяц, информации так нигде и не появилось, пока закрываю ишью, открою заново, если что-то прояснится

closed 2024-08-28 by u236