u236/homed-service-zigbee #198: Прошу добавить Energy meter.

NE8XT, 2024-08-01:
Description | _TZE204_81yrt3lo/TS0601
Logical type | router
Network address | 0x0a9d
Manufacturer code | 0x1141
Manufacturer name | _TZE204_81yrt3lo
Model name | TS0601

Permit join enabled successfully
Device "a4:c1:38:3a:6c:d9:5c:4d" joined network with address "0x0a9d"
Device "a4:c1:38:3a:6c:d9:5c:4d" interview started...
Device "a4:c1:38:3a:6c:d9:5c:4d" node descriptor received, manufacturer code is "0x1141" and logical type is "router"
Device "a4:c1:38:3a:6c:d9:5c:4d" active endpoints received: "0x01, 0xf2"
Device "a4:c1:38:3a:6c:d9:5c:4d" active endpoints received: "0x01, 0xf2"
Device "a4:c1:38:3a:6c:d9:5c:4d" endpoint "0x01" simple descriptor received
Device "a4:c1:38:3a:6c:d9:5c:4d" endpoint "0x01" simple descriptor received
Device "a4:c1:38:3a:6c:d9:5c:4d" endpoint "0xf2" simple descriptor received
Device "a4:c1:38:3a:6c:d9:5c:4d" endpoint "0xf2" simple descriptor received
Device "a4:c1:38:3a:6c:d9:5c:4d" manufacturer name is "_TZE204_81yrt3lo" and model name is "TS0601"
Device "a4:c1:38:3a:6c:d9:5c:4d" manufacturer name "_TZE204_81yrt3lo" and model name "TS0601" not found in library
Device "a4:c1:38:3a:6c:d9:5c:4d" identified as "_TZE204_81yrt3lo/TS0601"
Device "a4:c1:38:3a:6c:d9:5c:4d" interview finished successfully
Permit join disabled successfully

NE8XT, 2024-08-01:
[image]

u236, 2024-08-02:
https://github.com/u236/homed-service-common/commit/7222492f764747878d26238fe7e3be8800a98475 + https://github.com/u236/homed-service-zigbee/commit/460c497b6fadb6b51f0807601279461d18651459

closed 2024-08-02 by u236