u236/homed-service-zigbee #203: Прошу добавить поддержку Датчика движения SNZB-03P

bovops, 2024-08-25:
https://github.com/kirovilya/zigbee-herdsman-converters/blob/f63b540fdadb7f020df5bf6be8225a0893508c80/src/devices/sonoff.ts#L778

bovops, 2024-08-26:
upd: Попробовал добавить модель рядом с [MS-01](https://github.com/u236/homed-service-zigbee/blob/81245fcd05002c154641e6ab7318f511d61d94a2/deploy/data/usr/share/homed-zigbee/sonoff.json#L39), датчик всегда отдавал occupancy=true.

Похоже, проблема в отсутствии "motion_timeout", т.к. подцепил датчик в z2m, подвигал ползунок motion_timeout и датчик начал передавать разные значения occupancy в z2m.
Подцепил датчик обратно в homed-zigbee - снова всегда occupancy=true

bovops, 2024-08-26:
попробовал скопировать occupancyTimeout из SNZB-06P, получил ошибку:
Device "Датчик движения (тамбур)" occupancyTimeout action request failed, status code: "0xf0"

u236, 2024-08-28:
https://github.com/u236/homed-service-zigbee/commit/39552fed99886f5b18378610f7877056f2fd6649

> Device "Датчик движения (тамбур)" occupancyTimeout action request failed, status code: "0xf0"
это датчик с питанием от батарейки, я почти уверен, что для применения настроек его нужно "будить"

closed 2024-08-28 by u236