u236/homed-service-zigbee #236: Прошу добавить Реле SONOFF ZBMINIR2

alexaverdd, 2025-01-06:
{
  "active": true,
  "cloud": true,
  "description": "SONOFF/ZBMINIR2",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 2,
      "endpointId": 1,
      "inClusters": [
        0,
        3,
        4,
        6,
        2821,
        64599,
        64529
      ],
      "outClusters": [
        3,
        6,
        25
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "firmware": "1.0.4",
  "ieeeAddress": "1c:c0:89:ff:fe:6d:81:9a",
  "interviewFinished": true,
  "lastSeen": 1736196757,
  "linkQuality": 112,
  "logicalType": 1,
  "manufacturerCode": 4742,
  "manufacturerName": "SONOFF",
  "modelName": "ZBMINIR2",
  "networkAddress": 42848,
  "ota": {
    "currentVersion": 4100,
    "imageType": 8,
    "manufacturerCode": 4742,
    "running": false
  },
  "powerSource": 1,
  "supported": false,
  "version": 16,
  "name": "1c:c0:89:ff:fe:6d:81:9a"
}

u236, 2025-01-07:
https://github.com/u236/homed-service-zigbee/commit/6a0f50d927f85a8ea48ef95637c44e976f6cf104

alexaverdd, 2025-01-14:
Илья есть небольшая проблема, status не обновляется при нажатии на физический выключатель(тоесть если в вебе она была включена, а выключили физ выключателем, она осталась какбы включена в вебе , но на самом деле она выключилась) ,если переключать в веб интерфейсе все обновляется и выключается

u236, 2025-01-15:
> status не обновляется при нажатии на физический выключатель

Скорее всего дело в репортинге:
https://github.com/u236/homed-service-zigbee/commit/6b9b8b8eb6de2e38587adbdeb6fe00c01d6e6f36

Полсле обновления файла необходимо удалить реле из сети и добавить заново, чтобы оно повторно прошло интервью.

alexaverdd, 2025-02-24:
Илья, а можно ли сюда добавить ещё [External trigger mode (enum)](https://www.zigbee2mqtt.io/devices/ZBMINIR2.html#external-trigger-mode-enum)
External trigger mode, which can be one of edge, pulse, following(off), following(on).

u236, 2025-02-24:
> edge, pulse, following(off), following(on)

https://github.com/u236/homed-service-zigbee/commit/898e7d571cd928fdb7b44035443d8a96eebf3199