u236/homed-service-zigbee #239: PushOk Hardware/POK006 Садовый кран

gDIeMONd, 2025-01-09:
Прошу добавить!
[image]
`{
  "active": true,
  "cloud": false,
  "description": "PushOk Hardware/POK006",
  "discovery": false,
  "endpoints": [
    {
      "endpointId": 1,
      "inClusters": [
        0,
        3,
        6,
        1,
        18,
        17,
        20,
        19
      ],
      "outClusters": [
        0,
        25
      ],
      "profileId": 260
    }
  ],
  "ieeeAddress": "00:12:4b:00:2e:6c:24:fc",
  "interviewFinished": true,
  "lastSeen": 1736433058,
  "linkQuality": 252,
  "logicalType": 2,
  "manufacturerCode": 0,
  "manufacturerName": "PushOk Hardware",
  "modelName": "POK006",
  "networkAddress": 61915,
  "ota": {
    "currentVersion": 15,
    "imageType": 7657,
    "manufacturerCode": 22136,
    "running": false
  },
  "powerSource": 3,
  "supported": false,
  "name": "00:12:4b:00:2e:6c:24:fc"
}`

[конвертер](https://pushok.io/devices/pok006/pok006.js)

u236, 2025-01-09:
https://github.com/u236/homed-service-zigbee/commit/2ca5101aa7a710f988154474938d7245ee271f55

gDIeMONd, 2025-01-09:
valveStatus показывает цифры, а не фразы "closed", "open", "moving", "stuck"
забить?

u236, 2025-01-09:
опечатка закралась, поправил
https://github.com/u236/homed-service-zigbee/commit/72d6e45c06963d6b877206a5978b38e097914384