u236/homed-service-zigbee #240: _TZE204_dwcarsat/TS0601

Kexexe, 2025-01-15:
Просьба добавить датчик качества воздуха
_TZE204_dwcarsat/TS0601

Kexexe, 2025-01-15:
{
      "description":    "TUYA TS0601 Smart Air House Keeper",
      "modelNames":     ["_TZE200_dwcarsat", "_TZE204_dwcarsat"],
      "properties":     ["tuyaDataPoints"],
      "exposes":        ["temperature", "humidity", "co2", "voc", "formaldehyde", "pm25"],
      "options":        {
                          "tuyaDataPoints":
                          {
                            "2":    [{"name": "pm25", "type": "value"}],
                            "18":   [{"name": "temperature", "type": "value", "divider": 10}],
                            "19":   [{"name": "humidity", "type": "value", "divider": 10}],
                            "20":   [{"name": "formaldehyde", "type": "value", "divider": 1000}],
                            "21":   [{"name": "voc", "type": "value", "divider": 1000}],
                            "22":   [{"name": "co2", "type": "value"}]
                          },
                          "pm25": {"max": 1000}
                        }
    },

Kexexe, 2025-01-15:
Только сомневаюсь что воздух стал резко чище, проблема с запятой после ноля
ЛОВ было 14 стало 0,014 
Формальдегид было 4 стало 0,004
Остальное в норме.

Kexexe, 2025-01-15:
Понял метрику тут нужно поменять "divider": 1000}], на "divider": 10}], как у других.

Kexexe, 2025-01-15:
{
      "description":    "TUYA TS0601 Smart Air House Keeper",
      "modelNames":     ["_TZE204_dwcarsat"],
      "properties":     ["tuyaDataPoints"],
      "exposes":        ["temperature", "humidity", "co2", "voc", "formaldehyde", "pm25"],
      "options":        {
                          "tuyaDataPoints":
                          {
                            "2":    [{"name": "pm25", "type": "value"}],
                            "18":   [{"name": "temperature", "type": "value", "divider": 10}],
                            "19":   [{"name": "humidity", "type": "value", "divider": 10}],
                            "20":   [{"name": "formaldehyde", "type": "value", "divider": 10}],
                            "21":   [{"name": "voc", "type": "value", "divider": 10}],
                            "22":   [{"name": "co2", "type": "value"}]
                          },
                          "pm25": {"max": 1000}
                        }
    },
Рабочий конфиг.
Подозреваю что нужно объединить:
"modelNames":     ["_TZE200_mja3fuja", "_TZE200_dwcarsat", "_TZE204_dwcarsat"], должно быть, только запятая должна быть не на 1000 а на 10

u236, 2025-01-15:
https://github.com/u236/homed-service-zigbee/commit/9165951fcd4b0215e94a437489ee285203601978

closed 2025-01-15 by u236