u236/homed-service-zigbee #243: Добавление устройства.

xIMNIx, 2025-01-15:
Добрый день! 
Прошу добавить устройство.
Aqara Smart Wall Switch H1 (no neutral, double rocker) Модель: WS-EUK02

Manufacturer code | 0x115f
Manufacturer name | LUMI
Model name | lumi.switch.l2aeu1
Interview finished |  yes
Supported |  no
Active |  yes
Discovery |  yes
Cloud |  yes

(inf) zigbee:     Permit join enabled successfully
(inf) zigbee:     Device "1" rejoined network with address "0xb0f8"
(inf) zigbee:     Device "1" network address updated
(inf) zigbee:     Device "1" interview started...
(inf) zigbee:     Device "1" node descriptor received, manufacturer code is "0x115f" and logical type is "end device"
(inf) zigbee:     Device "1" active endpoints received: "0x01, 0x02, 0xf2"
(inf) zigbee:     Device "1" endpoint "0x01" simple descriptor received
(inf) zigbee:     Device "1" endpoint "0x02" simple descriptor received
(inf) zigbee:     Device "1" endpoint "0xf2" simple descriptor received
(inf) zigbee:     Device "1" manufacturer name is "LUMI" and model name is "lumi.switch.l2aeu1"
(wrn) zigbee:     Device "1" manufacturer name "LUMI" and model name "lumi.switch.l2aeu1" not found in library
(inf) zigbee:     Device "1" endpoint "0x01" properties recognized: temperature, status
(inf) zigbee:     Device "1" endpoint "0x01" actions recognized: status
(inf) zigbee:     Device "1" endpoint "0x01" exposes recognized: temperature, switch
(inf) zigbee:     Device "1" endpoint "0x02" properties recognized: status
(inf) zigbee:     Device "1" endpoint "0x02" actions recognized: status
(inf) zigbee:     Device "1" endpoint "0x02" exposes recognized: switch
(inf) zigbee:     Device "1" identified as "LUMI/lumi.switch.l2aeu1"
(inf) zigbee:     Device "1" endpoint "0x01" cluster "0x0002" binding to coordinator finished successfully
(inf) zigbee:     Device "1" endpoint "0x01" cluster "0x0006" binding to coordinator finished successfully
(inf) zigbee:     Device "1" endpoint "0x02" cluster "0x0006" binding to coordinator finished successfully
(inf) zigbee:     Device "1" endpoint "0x01" deviceTemperature reporting configuration request finished successfully
(wrn) zigbee:     Device "1" endpoint "0x01" cluster "0x0002" reporting configuration error, status code: "0x8c"
(inf) zigbee:     Device "1" endpoint "0x01" status reporting configuration request finished successfully
(wrn) zigbee:     Device "1" endpoint "0x01" cluster "0x0006" reporting configuration error, status code: "0x8d"
(inf) zigbee:     Device "1" endpoint "0x02" status reporting configuration request finished successfully
(inf) zigbee:     Device "1" interview finished successfully
(wrn) zigbee:     Device "1" endpoint "0x02" cluster "0x0006" reporting configuration error, status code: "0xc1"

Основная проблема - homed не получает статус второй кнопки при физическом нажатии.

u236, 2025-01-20:
https://github.com/u236/homed-service-zigbee/commit/49be663f7e3bc951836a6fdad660c5e3964cb045

closed 2025-01-20 by u236

xIMNIx, 2025-01-21:
Добрый день. 
После установки нового файла - "определился". 
Появились дополнительные кнопки в интерфейсе. 
Все кнопки из интерфейса вроде как работают. (Status memory - не  ясно)

Перестал получать данные от выключателя в Homed совсем. Не получает статус кнопок, не получает качество линка. На карте отображается отдельно от координатора. 
На нажатие физических кнопок выключателя Last seen не реагирует. 	

Network address	0x9685
Manufacturer code	0x1037
Manufacturer name	LUMI
Model name	lumi.switch.l2aeu1
Interview finished	
Supported	
Active	
Discovery	
Cloud	
Last seen	47 min
Exposes
Status left		on/off/toggle
Switch mode left	relay	decoupled/relay
Status right		on/off/toggle
Switch mode right	relay	decoupled/relay
Action left	•	
Action right	•	
Action both	•	
Temperature	•	
Indicator mode	default	default/inverted
Status memory	true	enable/disable
Link quality	192

u236, 2025-01-21:
status memory - запоминание состояния (восстановление после потери питания)
касательно остального - увы, мне особо нечего сказать, попробуй переджойнить выключатель, предварительным удалением, чтобы он заново прошел интервью

xIMNIx, 2025-01-21:
Пробовал и с удалением, потом перезагрузкой системы и поиском заново. Не спасает.
Может помогут какие-то логи, выгрузки? 
Может подскажете софт какой-то посмотреть, по сканировать? поискать что теряется?

u236, 2025-01-21:
не подскажу, к сожалению

покажи лог интервью и скажи какой используется координатор

xIMNIx, 2025-02-13:
Тест с другим координатором показал, что проблема в  координаторе zboss.
Спасибо!