u236/homed-service-zigbee #244: TUYA TS0202 Motion Sensor with Button _TZ3210_cwamkvua

Kexexe, 2025-01-15:
Description	TUYA TS0202 Motion Sensor with Button
Logical type	end device
Power source	
Network address	0xbc5d
Manufacturer code	0x1141
Manufacturer name	_TZ3210_cwamkvua
Model name	TS0202
Прошу добавить сущность датчика освещенности, в Z2M работает.

u236, 2025-01-15:
Датчик освещенности исправил: https://github.com/u236/homed-service-zigbee/commit/cf4f7abb07dc968745c2aa3e1a10400c6cf2a94f

Что касается кнопки - я уже миллион раз объяснял в чате, что состояния устройств публикуются в MQTT без флага retain и, фактически, "существуют" только в момент публикации. Поэтому "снимать статус" не имеет никакого смысла, это генерация лишнего, никому не нужного трафика.

Если речь про "зависание" статуса в HOMEd Web, то это просто кеш интрефейса, если обновить страницу, статусы снова станут неизвестными.

Если это проблема в контексте HA, есть целых два способа ее решения - [device trigger](https://www.home-assistant.io/integrations/device_trigger.mqtt/) и [event](https://www.home-assistant.io/integrations/event.mqtt/), и то и другое публикуется в MQTT Discovery для всех устройств со свойствами "action", "event" и "scene".

Менять поведение механизма публикации данных я не буду, это принципиальная позиция.

closed 2025-01-15 by u236

Kexexe, 2025-01-15:
Не отображает заряд батареи, пишет 0%
В Z2M писал 38,8%

u236, 2025-01-15:
> пишет 0%

0 это не "не отображает", 0 это 0

Kexexe, 2025-01-15:
Отображает неверное значение.

u236, 2025-01-15:
Не отображает ровным счетом ничего, кроме данных, полученных от устройства.