u236/homed-service-zigbee #249: ORVIBO Door and Window Sensor SM10ZW нет показаний состояния заряда батарейки

HassHomeKng, 2025-02-02:
{
  "active": true,
  "cloud": false,
  "description": "ORVIBO Door and Window Sensor SM10ZW",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 1026,
      "endpointId": 1,
      "inClusters": [
        0,
        3,
        1280,
        65535,
        1
      ],
      "outClusters": [
        0,
        4,
        3,
        5,
        1
      ],
      "profileId": 260,
      "zoneType": 21
    }
  ],
  "firmware": "v0.1.3",
  "ieeeAddress": "00:15:8d:00:01:ff:42:a0",
  "interviewFinished": true,
  "lastSeen": 1738457175,
  "linkQuality": 204,
  "logicalType": 2,
  "manufacturerCode": 4151,
  "manufacturerName": "ORVIBO",
  "modelName": "e70f96b3773a4c9283c6862dbafb6a99",
  "name": "Door Sensor",
  "networkAddress": 57125,
  "powerSource": 3,
  "supported": true,
  "version": 2
}

HassHomeKng, 2025-02-02:
Прошу добавить показания батарейки

[image]

u236, 2025-02-02:
должен отдавать, попробуй переджойнить с удалением, чтобы датчик заново прошел интервью, а вообще показания батарейки у некоторых датчиков могут передаваться раз в сутки или реже

closed 2025-02-02 by u236

HassHomeKng, 2025-02-02:
Я несколько раз его удалял (graceful и force) и добавлял.  Джойнил и к EZSP и к ZBOSS NCP, нигде нет данных заряда батрейки. Он у меня уже больше месяца я несколько раз его мучил и давал ему пару недель покоя, результата нет.

u236, 2025-02-03:
судя по всему, модель `e70f96b3773a4c9283c6862dbafb6a99` в принципе не отдает показания батарейки, по крайней мере в z2m для нее [нет](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6db3c99883d69099d5ba5a5805dada6ad02a2141/src/devices/livingwise.ts#L24-L31) этого параметра, а для других трех моделей [есть](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6db3c99883d69099d5ba5a5805dada6ad02a2141/src/devices/orvibo.ts#L89-L96), так что увы, я ничем не могу помочь

HassHomeKng, 2025-02-04:
Понятно. Спасибо! Надеюсь когда-нибудь исправят.