u236/homed-service-zigbee #251: Add to support 3-button switch Tuya

steshan-unix, 2025-02-06:
Hello. Please add to support this device

`{
  "active": true,
  "cloud": true,
  "description": "_TZE204_atpwqgml/TS0601",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 256,
      "endpointId": 1,
      "inClusters": [
        0,
        6,
        10
      ],
      "outClusters": [
        25
      ],
      "profileId": 260
    },
    {
      "deviceId": 256,
      "endpointId": 2,
      "inClusters": [
        6
      ],
      "profileId": 260
    },
    {
      "deviceId": 256,
      "endpointId": 3,
      "inClusters": [
        6
      ],
      "profileId": 260
    }
  ],
  "firmware": "0122052017",
  "ieeeAddress": "a4:c1:38:6a:13:f5:a6:5d",
  "interviewFinished": true,
  "lastSeen": 1738834199,
  "linkQuality": 112,
  "logicalType": 2,
  "manufacturerCode": 4742,
  "manufacturerName": "_TZE204_atpwqgml",
  "modelName": "TS0601",
  "name": "Сorridor",
  "networkAddress": 16604,
  "powerSource": 1,
  "supported": false,
  "version": 145
}`

yamaspb, 2025-02-06:
Не понял какие клавиши? На устройстве только одна кнопка для JOIN

[log.txt](https://github.com/user-attachments/files/18689997/log.txt)

u236, 2025-02-06:
> Не понял какие клавиши?

ты перепутал это ишью со своим, твое я прокомментировал и закрыл

u236, 2025-02-07:
https://github.com/u236/homed-service-zigbee/commit/c6116fd74759a2cc4bccd080d47405fc0d92b0ca

closed 2025-02-07 by u236

steshan-unix, 2025-02-07:
Прошу прощения ,за задержку, вот лог. Вдруг полезно будет

 Device "Сorridor" речь за него. По очереди было нажата каждая клавиша
 
[log.txt](https://github.com/user-attachments/files/18704271/log.txt)