u236/homed-service-zigbee #264: TUYA TS011F периодически отключается после обновления

cgtut, 2025-02-28:
[image]

[image]

обновился и стала периодически выключаться , переустановил заново HOMED (perenio)   всё равно какие-то ошибки, хотя включается и выключается
может просто совпало и розетка куку того , спасибо заранее

u236, 2025-03-01:
> обновился и стала периодически выключаться

что имеется в виду под словом "обновился"?

cgtut, 2025-03-01:
обновил homed web, homed zigbee, и тд, раньше стоял на homed zigbe ниже 3.8.5 ,

u236, 2025-03-01:
> обновил homed web, homed zigbee, и тд, раньше стоял на homed zigbe ниже 3.8.5 ,

тогда это скорее розетка умом тронулась, увы

closed 2025-03-01 by u236

cgtut, 2025-03-01:
скажите а как установить раннюю версию homed zigbee ? проверить на всякий случай

u236, 2025-03-01:
> скажите а как установить раннюю версию homed zigbee ? проверить на всякий случай

в архиве пакет 3.8.5 для переньки
[artifacts.zip](https://github.com/user-attachments/files/19037775/artifacts.zip)

гл я уверен, что проблема не в версии, эта долбаная `_TZ3000_okaz9tjs` есть у двух из трех пользователей HOMEd ZigBee, и я не исключение, разница только в версии прошивки

[image]

cgtut, 2025-03-01:
спс