u236/homed-service-zigbee #268: Датчик движения ORVIBO SN11

HassHomeKng, 2025-03-11:
[image]
[ORVIBO Electronics SN11 User Manual.pdf](https://github.com/user-attachments/files/19185475/ORVIBO.Electronics.SN11.User.Manual.pdf)

Прошу добавить датчик движения Orvibo SN11
После интервью распознаётся как датчик вибрации, и состояние сенсора не меняет.
{
  "active": true,
  "cloud": false,
  "description": "ORVIBO/895a2d80097f4ae2b2d40500d5e03dcc",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 1026,
      "endpointId": 1,
      "inClusters": [
        0,
        3,
        1280,
        65535,
        1
      ],
      "meta": {
        "zoneType": 45
      },
      "outClusters": [
        0,
        4,
        3,
        5,
        1
      ],
      "profileId": 260
    }
  ],
  "firmware": "v0.1.3",
  "ieeeAddress": "00:15:8d:00:01:95:e4:ea",
  "interviewFinished": true,
  "lastSeen": 1741701720,
  "linkQuality": 228,
  "logicalType": 2,
  "manufacturerCode": 4151,
  "manufacturerName": "ORVIBO",
  "modelName": "895a2d80097f4ae2b2d40500d5e03dcc",
  "networkAddress": 31552,
  "powerSource": 3,
  "supported": false,
  "version": 2,
  "name": "00:15:8d:00:01:95:e4:ea"
}

u236, 2025-03-11:
https://github.com/u236/homed-service-zigbee/commit/58ae1c0e16a3f7321bde1bd3deaed96479846b8b

closed 2025-03-11 by u236

HassHomeKng, 2025-03-11:
Спасибо!