u236/homed-service-zigbee #271: Moes BHT-006

admlaz, 2025-03-22:
Термостат теплого пола MOES-BHT006 (TS0601_TZE204_aoclfnxz). Проблема с получением и установкой параметров Temperature high и Hysteresis.

[image]

u236, 2025-03-22:
у меня таких три, с одинаковыми идентификаторами, но разными экранами, на двух работает, на одном нет, передавай привет туе, а я ничем не могу помочь

closed 2025-03-22 by u236

admlaz, 2025-03-22:
Понятно... Я с этим термостатом месяц мучаюсь. Начал с ZHA - там он только как роутер, нашел квирк - половинчатое решение. Вычитал, что в Z2M поддерживается, перевел всю сеть на него. Опять проблемы. Нашел кастомную прошивку, перепрошил - еще хуже. Прошил обратно на "родную" и снова накатил кастомную. Думал может плохо прошилось - не помогло. Снова прошил "родную" и вспомнил , что ставил homeD на попробовать. Подключил к нему - и, о чудо, все работает как часики, прокинул в HA - опять все отлично!!! За этим небольшим исключением. Но это фигня, деактивирую эти параметры в HA и все. Потестирую недельку, если будет вести себя так как сегодня -  лучшего и не надо! :) Спасибо!!! Прошивку родную взял в кавычки, ибо не уверен что родная. Нашел на том же сайте, что и кастомная. Шил с полным стиранием флэша, но вот номер версии прошивки смущает - он как у кастомной.