u236/homed-service-zigbee #272: Непонятная проблема с homeassistant discovery

admlaz, 2025-03-22:
[image]
На графике слева внизу видно, что после добавления термостата в HA, я пробовал им управлять вручную из HA и все нормально работало, т.е. hvac_action становилась heating при переходе термостата в режим нагрева(включение реле на термостате). Изменение всех режимов и температур отображались правильно как в homeD, так и в HA. В 23.05 сработала автоматизация в HA. На самом устройстве все нормально включилось и пошел нагрев. В homeD (слева в верху) - видим статус running: true, и правильные температуры. В HA, с температурами тоже все ОК, а вот текущий статус - Бездействие (hvac_action: "off", не очень понимаю почему не "idle").
В MQTT в топике термостата хоумасистатна вот что прописано:
{"action_template":"{{ \"off\" if value_json.running == false else \"heating\" if value_json.systemMode == \"heat\" else \"idle\" }}","action_topic":"homed/fd/zigbee/a4:c1:38:66:c0:a7:11:de","availability":[{"topic":"homed/device/zigbee/a4:c1:38:66:c0:a7:11:de","value_template":"{{ value_json.status }}"},{"topic":"homed/service/zigbee","value_template":"{{ value_json.status }}"}],"availability_mode":"all","current_temperature_template":"{{ value_json.temperature }}","current_temperature_topic":"homed/fd/zigbee/a4:c1:38:66:c0:a7:11:de","device":{"identifiers":["a4c13866c0a711de"],"model":"TUYA TS0601 MOES Electric Thermostat (Type A)","name":"moes-bht006","via_device":"homed-zigbee_homed"},"max_temp":35,"min_temp":5,"mode_command_template":"{\"systemMode\":\"{{ \"fan\" if value == \"fan_only\" else value }}\"}","mode_command_topic":"homed/td/zigbee/a4:c1:38:66:c0:a7:11:de","mode_state_template":"{{ \"fan_only\" if value_json.systemMode == \"fan\" else value_json.systemMode }}","mode_state_topic":"homed/fd/zigbee/a4:c1:38:66:c0:a7:11:de","modes":["off","heat"],"name":"Thermostat","preset_mode_command_template":"{\"operationMode\":\"{{ value }}\"}","preset_mode_command_topic":"homed/td/zigbee/a4:c1:38:66:c0:a7:11:de","preset_mode_state_topic":"homed/fd/zigbee/a4:c1:38:66:c0:a7:11:de","preset_mode_value_template":"{{ value_json.operationMode }}","preset_modes":["manual","program"],"temperature_command_template":"{\"targetTemperature\":{{ value }}}","temperature_command_topic":"homed/td/zigbee/a4:c1:38:66:c0:a7:11:de","temperature_state_template":"{{ value_json.targetTemperature }}","temperature_state_topic":"homed/fd/zigbee/a4:c1:38:66:c0:a7:11:de","unique_id":"a4c13866c0a711de_thermostat"}

admlaz, 2025-03-22:
Забыл добавить. В самом MQTT устройстве в HA я отключил(disable) сущности максимального нагрева и гистерезиса, которые не передавались ни в homeD, ни в HA от термостата. Не думаю что в этом причина, но уточнить обязан.

admlaz, 2025-03-22:
Блин, забыл рассказать как у меня все подключено...:)
MQTT и москито-брокер стоят на сервере HA. HomeD я для тестов на отдельный миникомпьютер установил под Armbian и Sоnof-E свисток. На нем москитоброкер не ставил, просто сразу указал в настройках HomeD сервер от HA. В конфигурации zigbee активировал HA discovery. Т.е. это не add-on HA.

u236, 2025-03-23:
```
"off" if value_json.running == false else "heating" if value_json.systemMode == "heat" else "idle"
```
все правильно вроде, не? :) idle будет в случае если systemMode не равно heat

admlaz, 2025-03-23:
Мне вообще не важно, Idle  будет или off в выключенном состоянии реле, я в своих автоматизациях, ну и HA, отслеживаю только running. Удивился, потому, что первый раз такое вижу. Перепробовал много разных решений по термостатам. Ни в одном не встречал для hvac_action названия состояния off. Обычно оно для hvac_modes используется. :) 
Я не программист, не могу по коду что-то подсказать, я сисадмин бывший, умею только анализировать и ставить задачи... :)
Если я правильно понял HomeD не публикует в сой топик текущее состояние hvac_action?

u236, 2025-03-23:
он публикует в свой топик свойство running

closed 2025-03-23 by u236

admlaz, 2025-03-23:
Опять же, если правильно понимаю, systemMode это hvac_modes. А оно может быть off, heat, cool, и т.д., а еще неизвестно и недоступно. hvac_action может быть только idle и heating, ну и неизвестно и недоступно.

u236, 2025-03-23:
я не понимаю смысла этой дискуссии

admlaz, 2025-03-23:
Да это не дискуссия..:) Просто пытаюсь объяснить... если не интересно - извините, и закройте топик. Не буду отнимать время.

u236, 2025-03-23:
список возможных состояний взят отсюда:
https://www.home-assistant.io/integrations/climate.mqtt/#action_topic

поменял местами `idle` и `off`, раз уж это так важно:
https://github.com/u236/homed-service-common/commit/69bbb48da4ee444040590141890d99c10e595134

admlaz, 2025-03-23:
Понятия не имею важно это или нет...:) Просто у них есть hvac_modes(System mode) - режим работы,  и hvac_action(Running) - текущее состояние исполнительного реле(не всего термостата). В первом - то будет, к примеру, heat, во-втором - heating. Второе от первого никак не зависит. Только если первое - off.

admlaz, 2025-03-23:
ИМХО, более правильным будет публикация HomeD состояния Running в свой топик и пнуть туда HA.

u236, 2025-03-23:
> ИМХО, более правильным будет публикация HomeD состояния Running в свой топик и пнуть туда HA.

а сейчас, по твоему, это как сделано?

admlaz, 2025-03-23:
ХЗ... я вижу какие-то условия, которые должен решать HA, вместо того, чтобы тупо взять значение из конкретного топика homeD.

u236, 2025-03-23:
> ХЗ... я вижу какие-то условия, которые должен решать HA, вместо того, чтобы тупо взять значение из конкретного топика homeD.

то есть, если я правильно понял, я должен взять, и перестроить нахер всю долбаную иерархию топиков и данных HOMEd, чтобы маленькому, бедному и несчастному HA, которым я даже не пользуюсь (ведь если бы он не был таким уродцем, то не было бы не какого HOMEd) не пришлось ничего решать?

серьезно?

u236, 2025-03-23:
пока ты не ответил, я предложу тебе альтернативный способ решения этой высосанной из пальца проблемы:

давай ты СНАЧАЛА будешь разбираться в сути вопроса, а уже ПОТОМ выносить мне пол дня этими бессмысленными разговорами? по-моему отличный план

admlaz, 2025-03-23:
нет, если homeD, публикует статус Running в кокой-то свой топик, где угодно, нужно просто HA объяснить где он находится, т.е. указать это в ha discovery. Извини, если чего недопонимаю, не программер я...

admlaz, 2025-03-23:
Я вроде давно предложил закрыть тему и извинился, ты сам продолжил...

u236, 2025-03-23:
ты сейчас издеваешься, или правда так сильно тупишь? то, о чем мы тут уже два часа говорим и есть дискавери, которое объясняет ха где брать состояние и как его понимать, ау, прием, алло

u236, 2025-03-23:
я даже ссылку на документацию скинул, ну же!

u236, 2025-03-23:
и я не продолжил, а просто сказал, что сделал небольшие правки