u236/homed-service-zigbee #274: Добавьте, пожалуйста, Aqara Roller Shade Driver E1

tpk23, 2025-03-26:
{
  "active": true,
  "cloud": true,
  "description": "LUMI/lumi.curtain.acn002",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 514,
      "endpointId": 1,
      "inClusters": [
        0,
        2,
        3,
        4,
        5,
        6,
        9,
        13,
        19,
        258
      ],
      "meta": {
        "position": 100
      },
      "outClusters": [
        10,
        25
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "ieeeAddress": "54:ef:44:10:00:a9:f9:51",
  "interviewFinished": true,
  "lastSeen": 1742979443,
  "linkQuality": 87,
  "logicalType": 2,
  "manufacturerCode": 4447,
  "manufacturerName": "LUMI",
  "modelName": "lumi.curtain.acn002",
  "networkAddress": 60668,
  "ota": {
    "currentVersion": 3615,
    "imageType": 14976,
    "manufacturerCode": 4447,
    "running": false
  },
  "powerSource": 0,
  "supported": false,
  "version": 31,
  "name": "54:ef:44:10:00:a9:f9:51"
}

u236, 2025-03-27:
для добавления дополнительных функций нужно устройство на руках, основные функции должны работать:
https://github.com/u236/homed-service-zigbee/commit/94e290d754d3d8fd3b35347c3dc2cfae3dea6c7c

closed 2025-03-27 by u236

tpk23, 2025-03-31:
Спасибо! Устройство добавилось. Работает только ползунок открывания/закрывания шторы в % и показывает температуру