u236/homed-service-zigbee #275: Поддержка протокола Ember для Zigbee 3.0 USB Dongle, координатор не видит раутеры через EZSP

alexandrustinov, 2025-03-29:
Проблема описана тут https://github.com/Koenkk/zigbee2mqtt/issues/15507

И решение тут https://github.com/Koenkk/zigbee2mqtt/issues/15507#issuecomment-1959359605

Если работать через deprecated протокол EZSP, то координатор подключает только устройства в прямой зоне видимости, а через раутеры игнорирует.
Аналогично картинка в web интерфейсе отображается некорректно - устройства или вяжутся линком с координатором, или и вовсе болтаются как-то отдельно, хотя это возможно другая проблема (см. рисунок ниже).

Иногда правда создается видимость работы через раутеры, т.е. устройства вне прямой видимости как-то комуницируют, возможно и через раутер, но подтвердить это у меня пока не получилось.

Проблема решается переходом на протокол Ember, но его homed не поддерживает.

Оборудование:

Coordinator:  Silicon Labs Model name	EZSP v13 Firmware	7.4.4-0
Router1: [TUYA TS011F Smart Plug with Power Monitoring](https://github.com/u236/homed-service-zigbee/blob/866166873fab1eabc90e3a85a70a11d86c91e5ff/deploy/data/usr/share/homed-zigbee/tuya.json#L1906)
Router2: [TUYA TS0001 or TS000F 1-Channel Relay with Power Monitoring](https://github.com/u236/homed-service-zigbee/blob/866166873fab1eabc90e3a85a70a11d86c91e5ff/deploy/data/usr/share/homed-zigbee/tuya.json#L1896)

alexandrustinov, 2025-03-29:
Картинка из homed-web - часть устройств висит в воздухе (EZSP), хотя связь с ними есть

[image]

zigbee2mqtt при этом через протокол ember рисует связи между раутерами нормально, ну и включение-выключение тоже работает. Похоже это проблема конкретной имплементации EZSP в этой прошивке координатора, но deprecated есть deprecated, работоспособность не гарантирована (переходите на Ember)

u236, 2025-03-29:
карта сети обновляется раз в час, с момента запуска службы, что касается багов конкретно версии ezsp - не могу предложить ничего лучше, чем просто поменять прошивку, ну а то что в z2m драйвер ezsp отмечен как deprecated и рекомендовано переходить на ember не имеет совершенно никакого отношения у homed-zigbee (на секундочку - ezsp это ember z-net serial protocol)

closed 2025-03-29 by u236