u236/homed-service-zigbee #276: homed-zigbee_3.9.7_mips_24kc.ipk перестал принимать данные со счетчика энергии

KonsGit, 2025-03-30:
homed-zigbee_3.9.7_mips_24kc.ipk перестал обновлять данные с датчика энергии.

[image]

Обновляется только last seen.

Откат на старую версию (3.6.2 нашел в чатик homed telegtam) возобновляет передачу данных (все работает).

На всякий случай прикрепляю zigbee log.

[homed.log](https://github.com/user-attachments/files/19525726/homed.log)

u236, 2025-03-30:
Не верю. В 3.9.7 была [упрощена](https://github.com/u236/homed-service-zigbee/commit/4c2288ae9c7643c6c99627e1305d2268c2742793) схема описания опций для tuyaDataPoints, и это коснулось не только этого счетчика, но и всех остальных устройств TS0601. Так что тут есть только два возможных варианта - либо ты каким-то образом используешь устаревший файл библиотеки, либо все остальные пользователи таких устройств партизаны, и не признаются, что у них что-то сломалось.

KonsGit, 2025-03-30:
На переньке все обновлено. Для этого датчика я написал конвертер с только тем, что мне нужно. Прикрепляю. Последний - он.

[TUYA.json](https://github.com/user-attachments/files/19526337/TUYA.json)

u236, 2025-03-30:
> [TUYA.json](https://github.com/user-attachments/files/19526337/TUYA.json)

Ну вот поэтому и перестало работать.  Повторяю - в 3.9.7 была упрощена схема описания опций tuyaDataPoints. Убери квадратные скобки так, чтобы
```
"111":   [{"name": "frequency", "type": "value", "divider": 100}]
```
превратилось в 
```
"111":   {"name": "frequency", "type": "value", "divider": 100}
```
и будь пожалуйста повнимательнее в другой раз.

closed 2025-03-30 by u236

KonsGit, 2025-03-30:
Спасибо за оперативность! 
В Homed web данные стали обновляться, но z2m не передаются.

[image]

u236, 2025-03-30:
Этот функционал сто лет не менялся, смотри топики, проверь что галочка discovery включена, что все кастомные expose правильно описаны в опциях и т.д. Для тестов можно попробовать убрать свой "конвертор" и проверить на "официальном". Больше я ничем не помогу.

KonsGit, 2025-03-30:
discovery была выключена. Все заработало. Спасибо!