u236/homed-service-zigbee #277: Датчик присутствия человека Tuya 24g _TZE284_iadro9bf/TS0601

larthe, 2025-04-03:
Два датчика от Tuya один 5.8 завелся второй на 24g нет.  _TZE284_iadro9bf/TS0601

24g 

TZE284_iadro9bf
{
  "active": true,
  "cloud": true,
  "description": "_TZE284_iadro9bf/TS0601",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 81,
      "endpointId": 1,
      "inClusters": [
        4,
        5,
        61184,
        0,
        60672
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "ieeeAddress": "a4:c1:38:09:80:86:08:7a",
  "interviewFinished": true,
  "lastSeen": 1743712573,
  "linkQuality": 134,
  "logicalType": 1,
  "manufacturerCode": 4417,
  "manufacturerName": "_TZE284_iadro9bf",
  "modelName": "TS0601",
  "name": "TZE284_iadro9bf",
  "neighbors": [
    {
      "linkQuality": 206,
      "networkAddress": 0
    },
    {
      "linkQuality": 97,
      "networkAddress": 64890
    }
  ],
  "networkAddress": 34630,
  "powerSource": 1,
  "supported": false,
  "version": 78
}

5.8g

{
  "active": true,
  "cloud": true,
  "description": "TUYA TS0601 Human Presence Sensor (Type G)",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 81,
      "endpointId": 1,
      "inClusters": [
        4,
        5,
        61184,
        0
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "ieeeAddress": "a4:c1:38:3d:32:1f:5c:40",
  "interviewFinished": true,
  "lastSeen": 1743712573,
  "linkQuality": 134,
  "logicalType": 1,
  "manufacturerCode": 4417,
  "manufacturerName": "_TZE204_laokfqwu",
  "modelName": "TS0601",
  "name": "Human Presence Sensor",
  "neighbors": [
    {
      "linkQuality": 157,
      "networkAddress": 0
    },
    {
      "linkQuality": 0,
      "networkAddress": 2110
    },
    {
      "linkQuality": 97,
      "networkAddress": 14882
    },
    {
      "linkQuality": 103,
      "networkAddress": 20236
    }
  ],
  "networkAddress": 64890,
  "powerSource": 1,
  "supported": true,
  "version": 74
}

[image]
[image]

larthe, 2025-04-03:
[image]

[image]

u236, 2025-04-04:
https://github.com/u236/homed-service-zigbee/commit/60869286e77c16afe74c81ba110fd6afa0f886ed

closed 2025-04-04 by u236

larthe, 2025-04-11:
@u236  К сожалению проблема с ним осталась, он стал поддерживаемым есть но, он не выдает значений =(( 
{
  "active": true,
  "cloud": true,
  "description": "TUYA TS0601 Human Presence Sensor (Type A)",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 81,
      "endpointId": 1,
      "groups": [
        1
      ],
      "inClusters": [
        4,
        5,
        61184,
        0,
        60672
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "ieeeAddress": "a4:c1:38:09:80:86:08:7a",
  "interviewFinished": true,
  "lastSeen": 1744386845,
  "linkQuality": 214,
  "logicalType": 1,
  "manufacturerCode": 4417,
  "manufacturerName": "_TZE284_iadro9bf",
  "modelName": "TS0601",
  "name": "Human Presence Sensor (Type A)",
  "neighbors": [
    {
      "linkQuality": 224,
      "networkAddress": 0
    },
    {
      "linkQuality": 54,
      "networkAddress": 64890
    }
  ],
  "networkAddress": 59584,
  "ota": {
    "currentVersion": 78,
    "imageType": 54179,
    "manufacturerCode": 4417,
    "running": false
  },
  "powerSource": 1,
  "supported": true,
  "version": 78
}

[image]

u236, 2025-04-13:
> он не выдает значений

я не знаю, чем помочь, без устройства на руках я могу только гадать о причинах