u236/homed-service-zigbee #280: Прозьба добавить устройство ZG-301Z.

tmclock, 2025-04-24:
Прозьба добавить устройство ZG-301Z.
https://sprut.ai/catalog/item/rele-tuya-smart-zigbee

u236, 2025-04-24:
я по ссылке на сайт спрута или по картинке на нем должен догадаться, что это за реле и как оно представляется в сети? :)

tmclock, 2025-04-24:
Где мне это можно посматреть? Нажать "#" и затем "Send". Или это гдето в другом месте смотреть?

tmclock, 2025-04-24:
{
  "active": true,
  "cloud": true,
  "discovery": true,
  "ieeeAddress": "70:b3:d5:2b:60:0e:b3:8c",
  "interviewFinished": false,
  "lastSeen": 1745489920,
  "linkQuality": 128,
  "logicalType": 1,
  "manufacturerCode": 4660,
  "name": "Свет Зал",
  "neighbors": [
    {
      "linkQuality": 0,
      "networkAddress": 18509
    },
    {
      "linkQuality": 160,
      "networkAddress": 39522
    }
  ],
  "networkAddress": 2350,
  "powerSource": 0,
  "supported": false,
  "version": 100
}

u236, 2025-04-24:
> "interviewFinished": false

интервью не пройдено, нет необходимой информации

u236, 2025-04-24:
> Что нужно зделать?

если адаптер ZBOSS и устройство не проходит интервью - смириться, в противном случае - заставить устройство пройти интервью

tmclock, 2025-04-24:
Понял. Отстал))) Спасибо.

tmclock, 2025-04-24:
Если я найду устройство. На нем пройдет интервью. Данные сюда сбросить и тогда будет и на ZBOSS рабоать?

u236, 2025-04-24:
> будет и на ZBOSS рабоать

увы, но нет

closed 2025-04-24 by u236