u236/homed-service-zigbee #282: Добавьте, пожалуйста, реле

tpk23, 2025-04-24:
{
  "active": true,
  "cloud": false,
  "description": "_TZE284_f5efvtbv/TS0601",
  "discovery": false,
  "endpoints": [
    {
      "deviceId": 81,
      "endpointId": 1,
      "inClusters": [
        4,
        5,
        61184,
        0,
        60672
      ],
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "ieeeAddress": "a4:c1:38:3d:f1:b1:77:84",
  "interviewFinished": true,
  "lastSeen": 1745400769,
  "linkQuality": 116,
  "logicalType": 1,
  "manufacturerCode": 4417,
  "manufacturerName": "_TZE284_f5efvtbv",
  "modelName": "TS0601",
  "name": "Реле 4",
  "neighbors": [
    {
      "linkQuality": 151,
      "networkAddress": 0
    },
    {
      "linkQuality": 42,
      "networkAddress": 23760
    },
    {
      "linkQuality": 88,
      "networkAddress": 24475
    },
    {
      "linkQuality": 75,
      "networkAddress": 57104
    }
  ],
  "networkAddress": 61982,
  "powerSource": 1,
  "supported": false,
  "version": 78
}

tpk23, 2025-04-24:
Девайс с четырьмя реле. Ни одоно из них не видно, управлять невозможно
Спасибо!)

u236, 2025-04-28:
https://github.com/u236/homed-service-zigbee/commit/2b12d5fdfa0283c1b3528a7cfce513cd00635690

closed 2025-04-28 by u236