u236/homed-service-zigbee #290: Просьба добавить 3-х фазную версию Bacchus Mercury Electricity Meter Controller

stigory, 2025-05-03:
Mercury_3ph
{
  "active": true,
  "cloud": false,
  "description": "Bacchus/Mercury_3ph_Counter",
  "discovery": true,
  "endpoints": [
    {
      "deviceId": 12,
      "endpointId": 1,
      "inClusters": [
        0
      ],
      "outClusters": [
        2820,
        1026
      ],
      "profileId": 260
    },
    {
      "deviceId": 12,
      "endpointId": 2,
      "inClusters": [
        1794
      ],
      "outClusters": [
        1794
      ],
      "profileId": 260
    }
  ],
  "firmware": "10/03/2024 23:44",
  "ieeeAddress": "00:12:4b:00:0c:c3:67:e0",
  "interviewFinished": true,
  "lastSeen": 1746232318,
  "linkQuality": 255,
  "logicalType": 1,
  "manufacturerCode": 0,
  "manufacturerName": "Bacchus",
  "modelName": "Mercury_3ph_Counter",
  "name": "Mercury_3ph",
  "neighbors": [
    {
      "linkQuality": 117,
      "networkAddress": 0
    },
    {
      "linkQuality": 143,
      "networkAddress": 14027
    },
    {
      "linkQuality": 118,
      "networkAddress": 14980
    },
    {
      "linkQuality": 57,
      "networkAddress": 51488
    }
  ],
  "networkAddress": 49005,
  "powerSource": 1,
  "supported": false,
  "version": 3
}

u236, 2025-05-11:
https://github.com/u236/homed-service-zigbee/commit/b413d1c94e496a44c4b9d3b80932bb780de05233

closed 2025-05-11 by u236

stigory, 2025-05-17:
Благодарю покорно.