u236/homed-service-zigbee #317: Slacky-DIY Tuya Thermostat with custom fw (model_r01)

n8065, 2025-09-13:
Пытаюсь описать это устройство, но пока без особого успеха.
Основная загвоздка в том, чтобы заставить работать scheduling, хотя и с парой других характеристик есть нюансы.

Девайс рассказывает о себе следующее:

```
{
  "active": true,
  "cloud": false,
  "description": "Tuya Thermostat for Floor Heating with custom Firmware (Type r01)",
  "discovery": false,
  "endpoints": [
    {
      "deviceId": 769,
      "endpointId": 1,
      "inClusters": [
        0,
        3,
        4,
        5,
        4096,
        8,
        513,
        516,
        514
      ],
      "meta": {
        "weekdayProgram": false
      },
      "outClusters": [
        25,
        10
      ],
      "profileId": 260
    },
    {
      "deviceId": 769,
      "endpointId": 2,
      "inClusters": [
        8
      ],
      "profileId": 260
    },
    {
      "deviceId": 97,
      "endpointId": 242,
      "outClusters": [
        33
      ],
      "profileId": 41440
    }
  ],
  "firmware": "v1.0.18",
  "ieeeAddress": "a4:c1:38:61:64:df:98:5a",
  "interviewFinished": true,
  "lastSeen": 1757763139,
  "linkQuality": 163,
  "logicalType": 1,
  "manufacturerCode": 25957,
  "manufacturerName": "Slacky-DIY",
  "modelName": "Tuya_Thermostat_r01",
  "networkAddress": 31243,
  "ota": {
    "currentVersion": 270020609,
    "fileName": "6565-0391-10183001-tuya_thermostat_zrd.zigbee",
    "fileVersion": 270020609,
    "imageType": 913,
    "manufacturerCode": 25957,
    "running": false
  },
  "powerSource": 1,
  "supported": true,
  "version": 16,
  "name": "a4:c1:38:61:64:df:98:5a"
}
```

На настоящий момент описан так:
```
{
  "Slacky-DIY": [
    {
      "description":    "Tuya Thermostat for Floor Heating with custom Firmware (Type r01)",
      "modelNames":     ["Tuya_Thermostat_r01"],
      "exposes":        [
                          "runningState", "thermostat", "childLock", "sensorType", "minHeatSetpointLimit", "maxHeatSetpointLimit", "temperatureOffset", "hysteresis", "deadZoneTemperature", "thermostatProgram"],
      "options":        {
                          "customAttributes": {
                            "minHeatSetpointLimit": {"type": "value", "clusterId": 513, "attributeId": 21, "dataType": 41, "divider": 100, "action": true},
                            "maxHeatSetpointLimit": {"type": "value", "clusterId": 513, "attributeId": 22, "dataType": 41, "divider": 100, "action": true},
                            "deadZoneTemperature": {"type": "value", "clusterId": 513, "attributeId": 25, "dataType": 40, "action": true},
                            "operationMode": {"type": "enum", "clusterId": 513, "attributeId": 37, "dataType": 24, "action": true},
                            "runningState": {"type": "enum", "clusterId": 513, "attributeId": 41, "dataType": 25, "action": true},
                            "sensorType": {"type": "enum", "clusterId": 513, "attributeId": 61440, "dataType": 48, "action": true}
                          },
                          "systemMode": {"enum": ["off", "heat"]},
                          "operationMode": {"enum": ["manual", "programming", "temporary"]},
                          "runningState": {"enum": ["idle", "heat"]},
                          "minHeatSetpointLimit": {"type": "number", "min": 5, "max": 45},
                          "maxHeatSetpointLimit": {"type": "number", "min": 5, "max": 45},
                          "deadZoneTemperature": {"type": "number", "min": 1, "max": 5, "step": 1},
                          "targetTemperature": {"min": 5, "max": 45, "step": 1},
                          "temperatureOffset": {"min": -9, "max": 9, "step": 0.1},
                          "hysteresis": {"min": 0, "max": 3, "step": 0.1},
                          "thermostatProgram": "moes",
                          "programTransitions": 4
                        }
    },
    {
      "modelNames":     ["Tuya_Thermostat_r01"],
      "properties":     ["thermostat", "childLock"],
      "actions":        ["thermostat", "childLock", "thermostatProgram"],
      "bindings":       ["thermostat"],
      "reportings":     ["thermostat"],
      "endpointId":     1
    }
  ]
}
```

В интерфейса выглядит как-то так:

[image]

Наверное, я весьма поверхностно понимаю код, но судя [по нему](https://github.com/u236/homed-service-zigbee/blob/master/actions/common.cpp#L124-L153), поддержка расписания, которое поддерживается этим устройством (3 вхождения: понедельник-пятница, суббота, воскресенье по 4 периода в каждом), пока не предусмотрено (там исключительно daily?).

При этом, для кастомных устройств (Moes/Tuya, например) я соответствующий код наблюдаю ("thermostatProgram": "moes", если не ошибаюсь), но его, понятное дело, сюда не пришьёшь. Если не прописываю опцию "thermostatProgram", то система кладёт и на опцию "programTransitions", рисуя в интерфейсе на каждое вхождение по 6 периодов.

Вопрос, собственно, в следующем: я что-то делаю не так, или пока нет смысла пытаться заставить это работать? Или, может быть, вопрос стоит адресовать к @slacky1965?

u236, 2025-09-13:
думаю стоит потерпеть, пока это сделает Федор, он уже разобрался более менее в структуре библиотеки, и у него в планах есть добавление термостатов

ишью пока подхожу открытым, напишу как будут новости

u236, 2026-01-02:
какой-то функционал кто-то добавил, по остальным вопросам предлагаю тыкать Федора

closed 2026-01-02 by u236

slacky1965, 2026-01-02:
> А вам, господа, спасибо за ваш труд, и с Новым годом!

Спасибо. Взаимно.

Ну и если будут вопросы, обращайтесь :))

u236, 2026-01-04:
@n8065 потегай меня в @homed_chat в телеге, помогу с расписанием

u236, 2026-01-05:
в песочнице, проверяй: https://github.com/u236/homed-service-zigbee/commit/421ece862152cf06ee31c1d57f4c210ac3f0ef65

n8065, 2026-01-05:
> в песочнице, проверяй: [421ece8](https://github.com/u236/homed-service-zigbee/commit/421ece862152cf06ee31c1d57f4c210ac3f0ef65)

Да, так лучше.

Заменил бинарник, выкинул термостат из сети, он перерегистрировался. Данные после перерегистрации не приехали.
После ввода значения "Weekday P1 Start" весь блок вычитался:

[image]